r-lidar/lidR #314: [2.2.x] Error in catalog processing

spono, 2020-01-15:
Ciao JR,
playing around with 2.2.0/.1 (as a result of #313 ) I'm not able anymore to process a catalog without getting the following:
`Error: i[1] is 1 which is out of range [1,nrow=0]`
Might it be related to wrong handling of empty chunks?

Jean-Romain, 2020-01-15:
Reproducible example please.

spono, 2020-01-15:
(sorry)
```
ctg = catalog(las.folder)
opt_chunk_size(ctg) <- 500
opt_chunk_buffer(ctg) <- 10

r = grid_canopy(ctg, res=0.5, p2r(0.1))
```

BTW, in the meantime I rolled back to 2.1.4  and it goes like a charm.

Jean-Romain, 2020-01-15:
As I mentioned, first install 2.2.0 from CRAN. `lidR` has now more than 900 unit tests that all run on on my machine, on travis CI and on different computers flavors on CRAN. Your example is trivial and should work. I'm pretty sure it comes from your installation. 

Also notice that `catalog` is deprecated (not formally yet) and that `chunk_buffer` is not respected by `grid_canopy` (see the documentation). But it does not matter actually this won't affect how it works.

spono, 2020-01-15:
yup, I still have to catch up with some changes..!

BTW, yes: the task is super-trivial and I tested both with 2.2.0 (from CRAN) and 2.2.1 (github) but I got that error. Tomorrow I can test more and reinstall everything

Jean-Romain, 2020-01-15:
Do you have troubleshooting with `grid_canopy` only or with all other functions?

spono, 2020-01-16:
The file is a strip few kms long and 80m wide.

this morning I tested:
- grid_canopy:
```
r = grid_canopy(ctg, res=0.5, p2r(0.1, na.fill=tin() ) )
Error: missing value where TRUE/FALSE needed
```

```
r = grid_canopy(ctg, res=0.5, p2r(0.1) )
i[1] is 1 which is out of range [1,nrow=0]
```

- grid_terrain:
```
r = grid_terrain(ctg, res=0.5, algorithm = knnidw(k=5L))
i[1] is 1 which is out of range [1,nrow=0]
Warning message:
There were 5 degenerated ground points. Some X Y Z coordinates were repeated. They were removed.
```

- grid_metrics:
```
r = grid_metrics(ctg, mean(Z), res=0.5)
i[1] is 1 which is out of range [1,nrow=0]
Warning messages:
1: Invalid data: RGB colors are recorded on 8 bits instead of 16 bits. 
2: Invalid data: RGB colors are recorded on 8 bits instead of 16 bits.
```

Same results with **2.2.0**(CRAN) and **2.2.1** (github).
I tried different options within each size range for:

- chunk size (30 -> ... -> 3000),
- resolution (0.5 -> ... -> 5)

I'm open to suggestions.

Jean-Romain, 2020-01-16:
I just run the following on a fresh install on lidR 2.2.0 on a virtual machine (W7). It worked smoothly

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = catalog(LASfile)
opt_chunk_size(ctg) <- 150
opt_chunk_buffer(ctg) <- 10
r = grid_canopy(ctg, res=0.5, p2r(0.1))
```

Jean-Romain, 2020-01-17:
No new of you. Did you fix your trouble?

spono, 2020-01-20:
> No new of you.

(unfortunately) you're right :)

> Did you fix your trouble?

No.
The example code you posted works fine for me as well, so this makes me think it might be an issue related to the data itself (I'm waiting for permissions before sending it). Nevertheless, it's pretty weird that the same process works smoothly on 2.1.4 [maybe something related to decimals or something wrong in the header? who knows..].

BTW, tomorrow I'll be able to test it on other machines and I'll let you have some news.

Jean-Romain, 2020-01-20:
No need to send the cloud for the moment. Save the catalog with `saveRDS` and share please.

Jean-Romain, 2020-01-21:
I was not able to reproduce with the archive you sent me. I asked you the catalog without the point cloud because if the error happens before any actual computation I can fake run the processing. This is what I did and it worked.

I have one concern anyway. `alt_dir` is missing in your catalog. Meaning that the object you saved has been build before 2.2.0 (see below). If you can reproduce this and it work, then I will need a true file and a minimal reproducible example. No need of the 4 files if only one is enough to fail. I have still no clue of what is your troubleshooting

```r
ctg = readRDS("ctg_file_W10_R362_v214.rds")
opt_chunk_size(ctg) <- 500
opt_chunk_buffer(ctg) <- 10

plot(ctg, chunk = TRUE) 
# error because alt_dir is missing. 
# It means the catalog has been build with < 2.2.0

# repair the catalog
ctg@input_options$alt_dir = ""

plot(ctg, chunk = T)

# works fine
lidR:::catalog_fakerun(ctg, perror = 0)
```

Jean-Romain, 2020-01-21:
I cannot reproduce only with the catalog. Without the point cloud I can only fake run the processing without actually computing anything.  This is what I did and it worked

```r
ctg = readRDS("/home/jr/Téléchargements/bug_314/ctg_test2_W10_R362_v220cran.rds")
opt_chunk_size(ctg) <- 500
opt_chunk_buffer(ctg) <- 10

plot(ctg, chunk = TRUE)

lidR:::catalog_fakerun(ctg, perror = 0)
```

If the code above works on your machine, I will need a point cloud and a fully (minimal) reproducible example. If you can reproduce with a single tile it 's better

Jean-Romain, 2020-01-21:
You did not told me that it append only in the 3rd chunk. So it was useless to test only with a catalog and without data. Now please make a complete minimal reproducible example that fails. An example in which I don't have to guess anything please...

spono, 2020-01-21:
it fails on the first empty chunk it finds: 3rd in the case of `chunk_size = 500`, 1st chunk in the case of `chunk_size = 50`.
BTW, I send everything asap

Jean-Romain, 2020-01-21:
So if I'm able to create an empty chunk I'm able to reproduce in theroy. Let me try first.

Jean-Romain, 2020-01-21:
Ok I reproduced easily. The next time please give me all the information the first time that will save us a lot of time. It comes from `readLAS` that fails at reading empty chunk for a reason I ignore. I will push a fix soon.

```r
chunk <- readRDS("/tmp/RtmpX2pUzz/chunk3.rds")
las <- readLAS(chunk)
#> Error in `[.data.table`(data, 1:.N, `:=`(X = round_any(X, factor), Y = round_any(Y,  : 
#> i[1] is 1 which is out of range [1,nrow=0] 
```

Jean-Romain, 2020-01-21:
Fixed

closed 2020-01-21 by Jean-Romain

spono, 2020-01-21:
ok, great! first guess was the correct one, then.
Sorry to make you lose time and thanks for the patience, as usual.

commit 1bc347f0 (2020-01-21): Fix #314