r-lidar/lidR #316: gap_fraction_profile not increasing upwards

LMurphy186232, 2020-01-23:
Thank you for a massively useful package. grid_metrics, in particular, is saving my life.

Bouvier et al defined gap fraction as "the ratio of the number of first returns below a
specified height threshold to the total number of first returns". I would thus expect the values to monotonically increase as I went up in the canopy, from near 0 to near 1. However, that is not what happens in my gap_fraction_profile results. I might get, in ascending height, gf values of 0.97, 0.92, 0.9, 0.94.

In digging into this, I suspect the reason is line 564 in utils_metrics.r which says `i = data.table::shift(cs)/cs`. Why the division by cs? (It is entirely possible I have missed some sort of correction factor that needs to be applied.)

Jean-Romain, 2020-01-23:
No, look at figure 1 in the paper. LAD is not monotonic. The equation 5 is incorrect. It does not match with the text. It is defined as the # of return in layer [0,z] divided by # of returns in layer [0,z+dz] which is what `data.table::shift(cs)/cs` does in a vectorized way.

It is easy to understand with 4 layers with 10, 5, 5 and 10 returns

1. LAD = 10/30 = 0.33
2. LAD = 5/20 = 0.25
3. LAD = 5/15 = 0.33
4. LAS = 10/10 = 1 (ground is always 1)

[0.33,0.25,0.33] is not monotonic. You can find the correct equation + some pictures in [this paper](https://www-sciencedirect-com.acces.bibl.ulaval.ca/science/article/pii/S0034425717305849).

closed 2020-01-23 by Jean-Romain

LMurphy186232, 2020-01-23:
Thank you for your reply! Would it be possible for you to write out the reference of the paper you are linking to in your reply? The link appears to go behind a password-secured area.

Jean-Romain, 2020-01-23:
Link edited + reference added.

LMurphy186232, 2020-01-23:
Perfect! Thank you for your time. I appreciate it.

Jean-Romain, 2020-01-23:
*Edit* LAD != gap fraction. In my answer you should read GF instead of LAD. The LAD is the log of GF divided by an coefficient of extinction.