r-lidar/lidR #317: Feature Request for L-moments and L-moment Ratios in .stdmetrics

redfoxgis, 2020-01-27:
Useful grid metrics for modeling forest attributes are L-moments and L-moment ratios. Currently, I am only aware of these metrics available in FUSION's grid metrics and cloud metrics functions. Are there any plans to include these in lidR's .stdmetrics?

Here is an paper showing the importance of L-moments in predicting forest fuels attributes (Figure 2):
https://www.mdpi.com/2571-6255/2/3/38

Here is the FUSION documentation which mentions use of L-moments:
http://forsys.cfr.washington.edu/Software/FUSION/FUSION_manual.pdf

Jean-Romain, 2020-01-27:
I did not even know about the term L-moment. Now I know. I guess one day or another I will have to create `.stdfusionmetrics`. It seems it is a recurrent request. But L-moment seems to have an untrivial definition:

https://github.com/cran/EnvStats/blob/master/R/lMoment.R#L1
https://github.com/cran/Lmoments/blob/master/R/Lmoments.R#L1

I guess I should take the [sample L-moment](https://en.wikipedia.org/wiki/L-moment#Sample_L-moments).

tgoodbody, 2021-05-06:
JR - I am looking to calculate these metrics as well. Judging by the rejected request tag I assume there is no implementation of this in lidR yet? May be worth another look as I am hearing about more users wanting to calculate L- metrics.

Jean-Romain, 2021-05-06:
The request was more than a year old and I did not find a single a day where I was motivated in working on that topic and it is not going to change. The request is good, simply I have many more thrilling feature to do so I gave up on this one. I don't know the which definition of L-moment I should use and I don't know which definition FUSION is using.

With external help the situation may change and I could do it.

tgoodbody, 2021-05-06:
Fair enough, sounds good.

tgoodbody, 2021-05-11:
Wow - thanks JR this is great... Appreciate you taking the time

rubenvalpue, 2021-05-12:
This is brilliant JR. Sorry, I just didn't know it was possible to do that, as I haven't worked that much with lidR. Definitely a working solution. I take back my 'critical miss' wording. I hope you didn't find it too harsh, but apologize if you did as it was not my intention. 
I agree you should keep dependencies to the minimum. They gave much headache after the last R version release, and they are prone to always give more. The more stand-alone the better.
Great work JR!

Jean-Romain, 2021-05-12:
No problem. I understand that there is a lot of documentation and people do not read every single document (but it is definitively documented in `grid_metrics()` and all other `metrics` related manual pages). I think it deserves an addition in the book somewhere in the "metrics¨ chapters. L-moment + fractal dimension as [suggested earlier](https://github.com/Jean-Romain/lidR/discussions/428) (I gave the same answer than here). @tgoodbody  any thought on the best place to put that?

rubenvalpue, 2021-05-12:
Thanks a lot Jean-Romain

tgoodbody, 2021-05-12:
JR - I will take a look now. I think it makes most sense in the `grid_metrics()` section. I need to review section 16 anyways :)

Jean-Romain, 2021-05-12:
Yes definitively. Those metrics have no meaning at other regularization level

rubenvalpue, 2021-05-12:
It may make sense to some in cloud_metrics too. Not for my stratification applications. But some people use them in the modelling too, like in that Mpdi paper on top of the thread. They are very stable moments, so they have been working particularly well at low densities. I think that's why.
Let me know if any help needed @tgoodbody

Jean-Romain, 2021-05-12:
Yes indeed. To me `cloud_metrics()` is only a convenient function used to develop models on plots but then metrics are truly computed with `grid_metrics()`. In that sense `cloud_metrics` is not really a tool in lidR. But it only my interpretation base on my usage.

We could put that in chap 8 which is short . Section [8.2 user defined metrics](https://jean-romain.github.io/lidRbook/metrics.html#metrics-user-defined) or in a new section 8.4 example of complex metrics using third party packages

tgoodbody, 2021-05-12:
Sounds good JR. I will write something up and push it later today for you to review.

tgoodbody, 2021-05-12:
@Jean-Romain I've created section 8.4 and added brief text about 3rd party package integration. I used the 2 examples discussed here (lmom and fractional dimensions) and gave some rationale for wanting to reduce the number of dependencies lidR has. Take a look at the edits I've pushed and let me know if you want anything changed.

Thanks @rubenvalpue for your comments and @redfoxgis for getting this conversation started! 👍

closed 2021-05-13 by Jean-Romain