r-lidar/lidR #319: writeLAS version selection?

jgrn307, 2020-02-06:
So I'm trying to visualize some data in the Mac version of cloudcompare which, unfortunately, only works with LAS 1.2 and below.  Is there any way to control writeLAS to select the LAS version I want (I assume it is defaulting to 1.4)?

Jean-Romain, 2020-02-06:
Please provide a reproducible example. `writeLAS()` does not have defaulting mode. It writes the format of the LAS object which is the format of the original file read with `readLAS()`. If you created a LAS object with `LAS()` without providing any header the format is likely to be 1.2

You can modify the format manually changing the header but if your LAS object is really a LAS 1.4 you could not easily write it in a LAS 1.2 without making other changes.

```r
las@header@PHB[["Version Minor"]] = 2L
```

Normally I no longer answer to questions about lidR by email or on github because I receive too many questions from too many users. If you have questions about using the lidR package please ask on [gis.stackexchange.com](https://gis.stackexchange.com/) with the tag `lidr`. On stackexchange.com other users can also answer your questions and the answers thus provided will be public and well indexed by search engines so they can also help others users.

jgrn307, 2020-02-06:
Sorry, the input is definitely a las 1.4 file -- this isn't a bug so much as a feature request -- add some ability to set LAS formats in writeLAS.  There's a reasonable number of external programs out there that can get grumpy with 1.4.  CloudCompare is an excellent visualization system, but on a Mac it it using an older verison of libLAS so there's no way to open a 1.4 file.

Jean-Romain, 2020-02-06:
`las2las` from LAStools is a free and open source tools that can convert your 1.4 into 1.2 (and the opposite) much efficiently that what could be done in R. See [this question](https://gis.stackexchange.com/questions/166130/converting-las-1-4-to-1-2). I don't see any incentive at making a copy of existing tools in less good. However you can convert manually in R if you really want to. Try something like:

```r
las@header@PHB[["Version Minor"]] = 2L
las@header@PHB[["Point Data Format ID"]] = 2 # Choose which prf fits with your data
las@data[["ScanAngleRank"]] <- as.integer(las@data[["ScanAngle"]]) # If prf >= 6
```

If you need more details please ask on gis.stackexchange and provide more information about your 1.4 data.

Regards

closed 2020-02-06 by Jean-Romain