r-lidar/lidR #322: Feature Request: Enhance lascheck function to list warnings and errors found during deep inspection in each las file of a catalog.

rs806, 2020-02-24:
Along the line of capturing and parsing output of lascheck from lidR (from question on stackexchange), could the lascheck function be enhanced so that it provides an output to aid in quality checking of las files by identifying problem files. thank you.

Jean-Romain, 2020-02-24:
Should `lascheck()` be enhanced this way? Yes. Could `lascheck()` be enhanced this way? Probably not without making many modifications. Fair feature request that I will try consider.

Jean-Romain, 2020-02-25:
In `lidR v3.0.0` `las_check()` ([formerly known as `lascheck()`](https://github.com/Jean-Romain/lidR/issues/321)) gains a new option `print = FALSE` and returns a `list` 

```r
las_check(las, FALSE)
#> $warnings
#> [1] "1 points are duplicated and share XYZ coordinates with other points"                                         
#> [2] "There were 1 degenerated ground points. Some X Y Z coordinates were repeated."                               
#> [3] "There were 1 degenerated ground points. Some X Y coordinates were repeated but with different Z coordinates."
#> 
#> $errors
#> [1] "Invalid header: X scale factors should be factor ten of 0.1 or 0.5 or 0.25 not 0.123"                      
#> [2] "Invalid file: the data contains a 'gpstime' attribute but point data format is not set to 1, 3, 6, 7 or 8."
```

closed 2020-02-25 by Jean-Romain

Jean-Romain, 2020-03-03:
In `v3.0.0` I also added `deep = TRUE` to run through each file of a LAScatalog and perform a deep inspection.