r-lidar/lidR #323: readLAS crashes when reading ALS las files, after previously working

RandCauthon, 2020-03-02:
I have previously been using the `lidR` package and it was working great (great work!). However today I simply tried to open an airborne las file and the R sessions crashed. Here is the code, it ain't complicated.

```r
library(lidR)
las = readLAS("Las_files/e1054n0839.las")
```

I have uploaded the example las file [here](https://drive.google.com/open?id=1EUn0jErmrM1a7za0k70pPa_MWuxMO8o1).

This crashes the R session, either in R or R studio (grey bomb). I have tried to open multiple airborne LiDAR las files from this same dataset and they all crash R. However, when I open much larger las files (100 times larger) from a different project they load into R just fine. I have updated my R version to 3.6.2, updated to the latest version of `lidR`, checked my filepaths, had the las file on my C drive, and on an external hard drive, doesn't matter. 

I have checked the las files in other programs such as PDQ viewer and ArcGIS and they look fine and perform fine in those software programs. And like I said everything was working fine yesterday.  I would also note that I had similar experience on a different computer, when I tried to `readLAS` or put into a las catalog the session crashed. I would be happy to learn that this behavior is due to something I, the user, is doing.

Jean-Romain, 2020-03-03:
I spotted the issue

```r
header <- rlas::read.lasheader("H:/Téléchargements/issue46/e1054n0839.las")
header <- lidR::LASheader(header)
proj4string <- lidR::projection(header, asText = FALSE)  # crash
```

Jean-Romain, 2020-03-03:
Minimal reproducible example

```r
wkt = "PROJCS[\"NAD83 (2011) / Conus Albers\",GEOGCS[\"GCS_NAD_1983_2011\",DATUM[\"D_NAD_1983_2011\",SPHEROID[\"GRS_1980\",6378137.0,298.257222101,AUTHORITY[\"EPSG\",7019]],AUTHORITY[\"EPSG\",1116]],PRIMEM[\"Greenwich\",0.0,AUTHORITY[\"EPSG\",8901]],UNIT[\"Degree\",0.0174532925199433,AUTHORITY[\"EPSG\",9102]],AUTHORITY[\"EPSG\",6318]],PROJECTION[\"Albers\",AUTHORITY[\"Esri\",43007]],PARAMETER[\"False_Easting\",0.0,AUTHORITY[\"Esri\",100001]],PARAMETER[\"False_Northing\",0.0,AUTHORITY[\"Esri\",100002]],PARAMETER[\"Central_Meridian\",-96.0,AUTHORITY[\"Esri\",100010]],PARAMETER[\"Standard_Parallel_1\",29.5,AUTHORITY[\"Esri\",100025]],PARAMETER[\"Standard_Parallel_2\",45.5,AUTHORITY[\"Esri\",100026]],PARAMETER[\"Latitude_Of_Origin\",23.0,AUTHORITY[\"Esri\",100021]],UNIT[\"Meter\",1.0,AUTHORITY[\"EPSG\",9001]]],VERTCS[\"NAVD_1988\",VDATUM[\"North_American_Vertical_Datum_1988\",AUTHORITY[\"EPSG\",5103]],PARAMETER[\"Vertical_Shift\",0.0,AUTHORITY[\"Esri\",100006]],PARAMETER[\"Direction\",1.0,AUTHORITY[\"Esri\",100007]],UNIT[\"Meter\",1.0,AUTHORITY[\"EPSG\",9001]],AUTHORITY[\"EPSG\",5703]]"
proj4 = rgdal::showP4(wkt)
crs = sp::CRS(proj4)  # crash
```

I'm closing the issue because it is not related to `lidR`

closed 2020-03-03 by Jean-Romain

Jean-Romain, 2020-03-05:
Fixed in v2.2.4

Jean-Romain, 2020-03-05:
For your information the conclusion is that the coordinate reference in your file is invalid. The issue went upstream to sp https://github.com/edzer/sp/issues/75 then upstream again to libproj https://github.com/OSGeo/PROJ/issues/2022

commit 4f78e917 (2020-03-04): Fix #323 + v2.2.4