r-lidar/lidR #324: Error using templated file name

azh2, 2020-03-03:
I am getting the following error when trying to use {ORIGINALFILENAME} in the catalog opt_output_file option
`# Internal engine will not return results into R. Instead it will write results in files.
opt_output_files(Tst1) <- "C:/Users/azh2/R/Out_Dir/{ORIGINALFILENAME}_DEM"

grid_terrain(las = Tst1, res = 1, algorithm = kriging(model = gstat::vgm(0.59, "Sph", 874), k = 10L),keep_lowest = TRUE)
`
`**Error in eval(parse(text = text, keep.source = FALSE), envir) : object 'ORIGINALFILENAME' not found**`

[image]

azh2, 2020-03-03:
The others appear to work fine, I am using {ID} now

Jean-Romain, 2020-03-03:
See https://github.com/Jean-Romain/lidR/issues/271

closed 2020-03-03 by Jean-Romain