r-lidar/lidR #325: Visualisation of point clouds with different colour palette

spono, 2020-03-10:
Ciao JR,
just for visualisation purposes, a potential improvement of #275 could be to have the possibility to plot together two pointclouds with different colour palette.

I think something like:
```
las1 # classified scene without trees
las2 # only trees
plot(las1, color="Classification") # mainly to distinguish ground VS manmade objects (buildings, power lines, etc)
plot(las2, add=TRUE) # add the trees coloured according to Z
```

Jean-Romain, 2020-03-10:
This is a fair feature request that can probably be done within a very short term.

Jean-Romain, 2020-03-10:
```r
library(lidR)
las = readLAS("~/Documents/ALS data/Fichiers las divers/Building/building_WilliamsAZ_Urban_normalized.las")
nonveg = filter_poi(las, Classification != LASHIGHVEGETATION)
veg = filter_poi(las, Classification == LASHIGHVEGETATION)

# Regular plot
plot(las, color = "Classification")

# Two point cloud with different palettes
x = plot(nonveg, color = "Classification")
plot(veg, add = x)
```

[image]

spono, 2020-03-10:
amazing!

closed 2020-03-10 by Jean-Romain