r-lidar/lidR #327: `lasrangecorrection()` and `sensor_tracking()` Errors

lucas-johnson, 2020-03-19:
Initially posted on GIS Stack Exchange: https://gis.stackexchange.com/questions/354527/lidr-range-correction-errors

The lastile used to produce the errors can be downloaded from the New York State GIS Clearinghouse FTP server: ftp://ftp.gis.ny.gov/elevation/LIDAR/NYSGPO_WarrenWashingtonEssex_2015/u_5355088400_2015.las

Jean-Romain, 2020-03-19:
I cannot access the file. It takes forever to connect the ftp and nothing happen.

lucas-johnson, 2020-03-19:
You can access the file via a web browser. You can go to this link: ftp://ftp.gis.ny.gov/elevation/LIDAR/NYSGPO_WarrenWashingtonEssex_2015/

and then download this file:
u_5370088700_2015.las

Jean-Romain, 2020-03-19:
No, as I said I have a Connection timed out

lucas-johnson, 2020-03-19:
Interesting... The file is pretty large (~200 MB ) so that probably has something to do with it. Do you have any suggestions for sending you the file? I tried compressing the file to `.laz`, but that file is 28 MB which exceeds github's 10 MB limit for files posted in issues.

Jean-Romain, 2020-03-19:
Google Drive, DropBox or any other file transfer service such as https://wetransfer.com/

lucas-johnson, 2020-03-19:
Sent via wetransfer...

Jean-Romain, 2020-03-20:
You have the most weirdly invalid file I've ever seen. The good new is that every invalid point is tagged. I don't know how to explain.

First look at 

```r
las = readLAS("u_5370088700_2015.las")
las = lasfilterduplicates(las)
fl236 = lasfilter(las, PointSourceID == 236)
plot(fl236, color = "gpstime")
```
It is a single flightline so the spatial order is the gpstime order. You should have a smooth gradient of color. But you can see some granularity in the gradient.

If we keep only the first and last return of multiple return we should have always two point per pulse

```r
# Get only the first and last returns of multiple returns
fl236fl <- lasfilter(fl236, (ReturnNumber == NumberOfReturns | ReturnNumber == 1) & NumberOfReturns > 1)
fl236fl <- laspulse(fl236fl)

# Should be 2 for each pulse because we kept only first and last return.
# Sometime it is 1 for some point on the very edge
tab = tabulate(fl236fl$pulseID)
incorrect = which(tab > 2)
```

We can see that after keeping only first and last return (2 points) we still have some pulses with 3 or more points. Let filter these invalid points

```r
invalid = lasfilter(fl236fl, pulseID %in% incorrect)
invalid@data[, .(X,Y,Z, gpstime, UserData,pulseID)]
#>               X       Y      Z   gpstime UserData pulseID
#>     1: 538486.9 4887087 493.93 115043503        0   11427
#>     2: 538481.9 4887087 478.89 115043503        0   11427
#>     3: 538497.9 4887000 480.84 115043503        1   11427
#>     4: 538485.8 4887089 493.67 115043503        0   11449
#>     5: 538481.0 4887088 478.82 115043503        0   11449
#>   ---                                                   
#>199608: 538414.4 4888414 531.37 115043522        1  183387
#>199609: 538409.5 4888499 554.51 115043522        0  183391
#>199610: 538407.4 4888499 549.00 115043522        0  183391
#>199611: 538420.6 4888414 547.65 115043522        1  183391
#>199612: 538414.7 4888414 531.45 115043522        1  183391
```

We can see that `UserData` is set to 1 for many of them. Here is the trouble. Points tagged `UserData` are invalid points with a weird gpstime. Look at the following. The color gradient is smooth

```r
las = readLAS("~/Téléchargements/u_5370088700_2015.las")
las = lasfilterduplicates(las)
las = lasfilter(las, UserData == 0)
fl236 = lasfilter(las, PointSourceID == 236)
plot(fl236, color = "gpstime")
```

And the sensor tracking runs well

```r
sensor = sensor_tracking(las)
plot(las) %>% add_flightlines3d(sensor)
```

Jean-Romain, 2020-03-20:
`sensor_tracking()` know handle your case

```r
sensor_tracking(las)
#> Err0r : After keeping only first and last returns of multiple returns pulses, 7991 pulses still have more than 2 points. This dataset is corrupted and gpstime is likely to be invalid. 
```
I also updated `lascheck()` to add a test of gpstime coherence. Sadly this test takes a lot of time and slow down the check. I will try to improve that.

```
 Checking the data
  - Checking coordinates... ✓
  - Checking coordinates type... ✓
  - Checking attributes type... ✓
  - Checking ReturnNumber validity... ✓
  - Checking NumberOfReturns validity... ✓
  - Checking ReturnNumber vs. NumberOfReturns... ✓
  - Checking RGB validity... ✓
  - Checking absence of NAs... ✓
  - Checking duplicated points...
   ⚠ 1474 points are duplicated and share XYZ coordinates with other points
  - Checking degenerated ground points...
   ⚠ There were 4 degenerated ground points. Some X Y Z coordinates were repeated.
  - Checking attribute population... ✓
  - Checking gpstime incoherances
    ✗ 1639365 pulses (points with the same gpstime) have points with identical ReturnNumber
  - Checking flag attributes... ✓
  - Checking user data attribute...
   ⚠ 3266698 points have a non 0 UserData attribute. This probably has a meaning.
 Checking the header
  - Checking header completeness... ✓
  - Checking scale factor validity... ✓
  - Checking point data format ID validity... ✓
  - Checking extra bytes attributes validity... ✓
  - Checking coordinate reference sytem... ✓
 Checking header vs data adequacy
  - Checking attributes vs. point format... ✓
  - Checking header bbox vs. actual content... ✓
  - Checking header number of points vs. actual content... ✓
  - Checking header return number vs. actual content... ✓
 Checking preprocessing already done 
  - Checking ground classification... yes
  - Checking normalization... no
  - Checking negative outliers... ✓
  - Checking flightline classification... yes
```

Jean-Romain, 2020-03-20:
Tomorow I'll investigate on `lasrangecorrection()`

Jean-Romain, 2020-03-20:
For `lasrangecorrection()` I confirm it was a bug with a bad matching at the very edge of a flightline. Hopefully I added safety guards for very weird results.

You must know that the computation is necessarily weaker at the edges of the point cloud because of the lower number of points used to estimate the sensor positionning. Before to compute the range correction check if the positions look good and well aligned especially on the edges. But the good way is to use a large buffer to compute the sensor positions.

Jean-Romain, 2020-03-20:
I created a fast version of the gpstime coherence test. Issue closed.

closed 2020-03-20 by Jean-Romain

lucas-johnson, 2020-03-20:
@Jean-Romain Thank you for investigating this and taking care of it so quickly. This is super helpful. I would like to propose a feature that would help to determine if the intensity values require range-correction. Plotting something like the following (taken from http://www.geogra.uah.es/images/Documentos/emilio/PDF/garcia2010.pdf) would be helpful to diagnose if the intensity values have already been range corrected: 
[image]

 I'm going to try and produce this myself with the help of the output `sensor_tracking()`

Jean-Romain, 2020-03-20:
If you need a function that return the range of each point ask me. This function does not exist because the range is discarded after intensity normalization. But it is few lines of code to change to return the range of each point. Do not try it yourself I already spent a lot of time thinking about special cases (and yet I missed one that was the source of your bug).

lucas-johnson, 2020-03-20:
@Jean-Romain Ok - yes it would be very helpful to have a function that computes the range of each point.  I'm happy to create a feature request issue if you would like.

Jean-Romain, 2020-03-20:
Yes please open a feature request so I can keep tidy logs of what I'm doing. Thanks

lucas-johnson, 2020-03-20:
Feature request submitted: https://github.com/Jean-Romain/lidR/issues/328