r-lidar/lidR #329: [2.2.3] Different filter behaviour in lasclip

spono, 2020-03-22:
The `filter` option in `lasclip` behaves differently if used with `LAS` and `LAScatalog` objects: in the first case it doesn't have any effect (but does not return an error either) and requires a second step using `lasfilter`, while in the second one it works perfectly. Is it meant to behave like this?
If so, I think it might be a possible enhancement or, at least, be the case of adding a warning or info in the Help page.

Below a reproducible example:

```
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")

## LAScatalog

ctg = readLAScatalog(LASfile)
opt_chunk_size(ctg) <- 150
opt_chunk_buffer(ctg) <- 10
opt_filter(ctg) <- '-drop_class 2'

tmp = lasclipCircle(ctg, 684870.0,5017796.9, 2)
unique(tmp@data$Classification)

#> [1] 1

## LAS
las = readLAS(LASfile)
unique(las@data$Classification)

#> [1] 2 1

tmp = lasclipCircle(las, 684870.0,5017796.9, 2, filter= '-drop_class 2')
unique(tmp@data$Classification)

#> [1] 2 1

tmp = lasclipCircle(las, 684870.0,5017796.9, 2, filter= ~Classification!=2)
unique(tmp@data$Classification)

#> [1] 2 1 
```

I guess filters such `'-drop_class 2'` work only at `LASlib` level when reading from file, right?
Thanks!

Jean-Romain, 2020-03-22:
There is no argument `filter` in `lasclip()`. See the documentation. Also `filter = "-string"` is something that (when existing) only works **at read time**'.

spono, 2020-03-22:
yeah, true, in the case of a catalog the filtering is done at `readLAScatalog` level, and hence what arrives to `lasclip` is already "clean". My bad.

Excluding the input functions that accept `filter = "-string"`, what's the rationale behind the fact that some functions have a `filter` option (e.g. point_metrics, grid_metrics, etc) and others don't? Complex implementation/lack of scripting time or something else?

Jean-Romain, 2020-03-23:
Both + some rationale. 

You can always replace `filter = ~expression` by `lasfilter` + `function()` + optionally `rbind` is you want to merge back two split point clouds. But this is not clever in term of memory usage and computation speed. The goal of the argument `filter` is to provide a memory efficient equivalent. 

Let me give you and example using `lasfiltershape` because it was the first function to get an argument `filter`

Imagine you want to detect flat surfaces but to go faster you don't want to process the ground points. Moreover we will assume that your dataset is 500 MB and 10% of the points are ground. 

```r
las = readLAS() # 500 MB
notgnd = lasfilter(las, Classification != 2) # +450 MB
notgnd = lasdetectshape() # + approx 15 MB
```
You need 450 MB of extra memory to do that (almost twice the memory). And yet is is not finished because now you have an object `notgnd` which is missing the ground points. We can add those points

```r
gnd = lasfilter(las, Classification -= 2) # +50 MB
las = rbind(gnd, notgnd) # + 500 MB
```

Great! To process your 500 MB object you need 1.5 GB of extra memory allocated yourself + the memory allocated by `lasdetectshape` itself.

Hopefully `lasdetectshape` has an argument `filter` that allow for doing the same for 0 MB of memory

```r
las = lasdetectectshape(, filter = ~Classification != 2)
```

Now you understand why there is a `filter` argument in some functions you need to understand why some functions do not have such options. Several answers:

**1. Because it is almost useless**

For example 

 ```r
las2 = lasclip(las, ..., filter = ~Classification != 2)
```

Can be replaced by

```r
ctg = readLAScatalog(..., filter =  "-drop_class 2")
las2 = lasclip(ctg, ...)
```

And it is much better. Also adding `filter` would be very confusing with `opt_filter()` with inconstant behaviors.

**2. It is not possible**

`filter` is a way to be fast and memory efficient. I means that under the hood there is necessarily some C++ involved. Memory management in R is virtually not existent. So it means the the internal must be based on my own C++ code. But sometime I'm coding either in R or I'm using third party packages with no such optimization.

**3. It is to hard**

Sometime it just technically hard.

I hope this answer your question.

spono, 2020-03-23:
> 1. Because it is almost useless

Well, if it helps memory efficiency while working on memory-loaded clouds (like e.g. point_metrics), I wouldn't consider it almost useless but yes: I understand that, balancing pros&cons, it can't be thought for all the functions.

I imagined it was a mix of reasons, but I was curious to understand a bit more the idea behind some choices. So, yes: it definitely answers my question!
thank you very much, as usual.

Jean-Romain, 2020-03-23:
> Well, if it helps memory efficiency while working on memory-loaded clouds I wouldn't consider it almost useless

But it is not always the case. The example of `lasclip` is good because `lasclip` is purely R based. I cannot do much better than the user. If I add an option `filter` the only thing I can do is adding `lasfilter` inside the function after the clipping. Exactly like you but you will add `lasfilter` after the function. Not a big difference. So yes, useless because it is not more memory efficient.

Well actually it could be a little be more efficient in some specific cases. But I also have to balance with the fact that all functions must be used consistently used with a LAS and LAScatalog. Look carefully `filter` exists only for functions that do not support a LASctalog. So not clash or confusion between `opt_filter() <- "-string"` and `filter = ~formula`. There is one exeption which is `grid_metrics()` but with the insight I should not have added that in `grid_metrics()`... too late.

closed 2020-03-23 by Jean-Romain