r-lidar/lidR #330: Simple Range-Correction Alternative

lucas-johnson, 2020-03-23:
## Feature Request / Enhancement

### Current Behavior
I have been having a hard time using the `sensor_tracking()` and `lasrangecorrection()` functions that are currently offered (See here https://github.com/Jean-Romain/lidR/issues/327 and here https://gis.stackexchange.com/questions/354769/split-las-data-from-multi-beam-sensors). My difficulties have been mostly due to the intricacies of the las datasets that I am working with. The current  workflow requires perfectly populated `gpstime`, `NumberOfReturns`, `ReturnNumber`, and `PointSourceID` to produce flight lines, and then use the flight lines to range-correct intensity values. These requirements are too strict for my datasets. 

### Proposed Alternative
It might be useful to have a less stringent and perhaps less precise method for range correction. Perhaps the user may supply an average sensor elevation and a grid resolution. The method can then compute a range based on difference between median las height at a specified gridded resolution and the average sensor height.  This kind of method is outlined in this paper: https://www.sciencedirect.com/science/article/abs/pii/S0034425707001812

Jean-Romain, 2020-03-23:
You problem is very specific and according to Martin Isenburg's blog post you previously mentioned it is even more specific than that because data providers are now expected to provide valid files even with this kind of sensor.

What you are suggesting here is to get rid of the XYZ sensor coordinates computed with `sensor_tracking()` and instead fead the function with a fixed sensor elevation and no XY sensor coordinates.  With no XY so we can't take into account the increasing path due to scan angle but it is fair enough because we are looking for a less accurate method. Also with a fixed Z and no XY the range becomes `Zsensor - Zpoint`

If this is what you are looking for I'm sorry to say that I won't implement such feature because you can do it in 2 lines of code.

```r
zsensor = 2200
range = zsensor - las$Z
```

If you want a normalized intensity one more line is sufficient.

```r
Intensity = las$Intensity * (range/Zsensor)^2.3
```

Also I won't add such feature because it is very weak. Below an image from a paper under review. It is two intensity maps before and after range correction. You can see that the sensor was definitely not at a constant elevation for each flightlines.

[image]

lucas-johnson, 2020-03-23:
@Jean-Romain Ok thanks for taking the time to look this over. Thanks for explaining how weak this option is, and how simple it would be to compute.

closed 2020-03-23 by Jean-Romain