r-lidar/lidR #331: Mismatch

cyril06, 2020-03-24:
Hi,
I'm working for a LIDAR integrator and I'm charge of evaluating the strip adjustment algorithm we've developped. 
I'd like to know if it is possible to evaluate the mismatch in lidR. And if yes how would you process.

For instance in TerraSolid there is a "Measure match" tool. But I'd like to generate some graphs showing the benefif of strip adjustment.

If you have any idea.

Take care
Cyril

Jean-Romain, 2020-03-24:
Please clarify the question because I don't understand what you are expecting. Are you asking me if there are tools in lidR to validate your strip adjustment algorithm? Are you asking me if there is a strip adjustment algorithm? (Answer is no).

Jean-Romain, 2020-03-24:
Question duplicted here https://gis.stackexchange.com/questions/354980/evaluate-mismatch-in-lidr. Closing.

closed 2020-03-24 by Jean-Romain

cyril06, 2020-03-24:
I'm not asking if there is a strip adjustment algorithm in lidR but how you would process to evaluate it. But rather I'd like to know if there is a mismatch algorithm for instance.
Or if it could be possible to evaluate the mismatch between two or more flightlines by calculating the difference in Z on a flat surface.
In fact I'm trying to think of graphs showing before/after the strip adjustment. And by that showing the interest of strip adjustment. If you have any idea it would be nice.

Cyril