r-lidar/lidR #332: Error when running `grid_terrain()`

lucas-johnson, 2020-03-25:
Running grid terrain on a catalog with the following settings:
```
opt_stop_early(ctg) <- FALSE
opt_chunk_buffer(ctg) <- 100
opt_filter(ctg) <- "-drop_withheld -drop_class 7 18"
opt_output_files(ctg) <- "~/Downloads/dtm/{*}"
opt_progress(ctg) <- FALSE
```
I have also selected some files as buffer only:
[image]

I get the following error message:
```
> grid_terrain(ctg, res=1, knnidw(k=5))
The original tiling pattern does not match with the resolution 1. Chunks were extended to avoid partial pixels.
Error in if (all(!process[tile_is_in_bbox])) select <- FALSE else if (sum(tile_is_in_bbox) >  : 
  missing value where TRUE/FALSE needed
```

Jean-Romain, 2020-03-25:
This is not related to `grid_terrain` but to the function that creates the chunks. Please save your `LAScatalog` with `saveRDS` and share it. This should be enought to reproduce. Please also mention your `lidR` version.

lucas-johnson, 2020-03-25:
Yeah it seems to be a problem with grid_metrics as well.  I encountered the bug with the `devel` branch (lidR 3.0.0), but have sense reinstalled the current cran distribution (lidR 2.2.3), and have the same issue. 

[ctg_chunk_bug.rds](https://github.com/Jean-Romain/lidR/files/4382046/ctg_chunk_bug.rds)

lucas-johnson, 2020-03-25:
Seems to be working in version 2.2.2

Jean-Romain, 2020-03-25:
Could you confirm if this work or not

```r
lidR:::catalog_makecluster(ctg)
```

Surprisingly it works on my computer

lucas-johnson, 2020-03-25:
Yes it works in version 2.2.3.

lucas-johnson, 2020-03-25:
R

Jean-Romain, 2020-03-25:
Ok I reproduced. It fails with the partial processing only.

```r
library(lidR)
ctg = readRDS("~/Téléchargements/ctg_chunk_bug.rds")
lidR:::catalog_makecluster(ctg, realignment = list(res = 20, start = c(0,0))) # fail

ctg$processed = NULL
lidR:::catalog_makecluster(ctg, realignment = list(res = 20, start = c(0,0))) # ok
```

Jean-Romain, 2020-03-25:
Fixed

closed 2020-03-25 by Jean-Romain

lucas-johnson, 2020-03-25:
Thanks so much!

commit b012628f (2020-03-25): Fix #332