r-lidar/lidR #336: lasrangecorrection() does not work with LAScatalog

lucas-johnson, 2020-03-27:
The [man pages](https://rdrr.io/cran/lidR/man/lasrangecorrection.html) indicate that `lasrangecorrection()` can take `LAScatalog` but the behavior indicates otherwise:

```
> test <- readLAScatalog("~/Downloads/huntington_tiles_raw/")
> test_corrected <- lasrangecorrection(test, sensor, 2000)
Error in UseMethod("normalize_intensity", las) : 
  no applicable method for 'normalize_intensity' applied to an object of class "c('LAScatalog', 'SpatialPolygonsDataFrame', 'SpatialPolygons', 'Spatial', 'SpatialVector')"
```

Jean-Romain, 2020-03-27:
It is a documentation mistake (bad copy paste). Functions that actually support a LAScatalog have two more section named **Working with a LAScatalog** and **Supported processing options**.

`lasrangecorrection` does not support `LAScatalog` because it is still pretty young and bug prone (as you experienced). But it is straightforward to implement. Simply use [lasnormalize](https://github.com/Jean-Romain/lidR/blob/master/R/lasnormalize.r#L187) as template.

lucas-johnson, 2020-03-27:
Ok. I tried to make my own version with `catalog_apply()`, but I'm hitting issues. Unrealistic ranges are being computed:
```
An high range R has been computed relatively to the expected average range Rm = 2061
Point number 1 at (x,y,z,t) = (555019.24, 4876499.69, 541.34, 114989437.22)
Matched with sensor between (566610.26, 4869809.93, 2644.10, 114989437.00) and (566609.23, 4869882.60, 2644.43, 114989437.50)
The range computed was R = 13530.88
Check the correctness of the sensor positions and the correctness of the gpstime either in the point cloud or in the sensor positions.
```

I have plotted the catalog, and the tile being processed (red), as well as my sensor positions (black), and the sensor points for which point number 1 is being matched (green). It is quite far off...
[image]

Jean-Romain, 2020-03-27:
> It seems strange that sensor_tracking() works with a catalog, but lasrangecorrection() doesn't.

Because I felt that `sensor_tracking()` was a robust function but not `lasrangecorrection()`.  And I was right. Moreover to be robust `lasrangecorrection()` must use sensor position that encompass more than the tile. An input that can only be produced if `sensor_tracking()` works with a `LAScatalog`.

Considering the error message the matching is correct. Your point #1 has a gpstime `t = 114989437.22`. The position of the sensor was interpolated between `t = 114989437.00` and `t = 114989437.50` which is correct. Sadly it appears that those two sensor locations are not in the good tile. Why? I don't know. The simplest guess is that gpstime times are duplicated and are not consistent across tiles. Try to plot `spplot(sensor, "gpstime")`

lucas-johnson, 2020-03-27:
To add some detail, I computed the sensor positions separately for `UserData == 1` and `UserData == 0` (due to my [multi-beam problem](https://gis.stackexchange.com/questions/354769/split-las-data-from-multi-beam-sensors)). And then merged the two SPDFs together. If I try `lasrangecorrection()` with one set of sensor positions and one set of points, the problem still occurs. 

[image]

lucas-johnson, 2020-03-27:
plot from first set of sensor positions: 
[image]

lucas-johnson, 2020-03-27:
plot from second set of sensor positions: 
[image]

Jean-Romain, 2020-03-27:
I don't understand. The first set looks good, the second set looks good. The gpstime seems to match but if you merge both sets you have a weird pattern.

lucas-johnson, 2020-03-27:
Yes very odd. I recomputed the two sets of sensor positions, and remerged them, and now it looks good:
```
merged_sensors <- rbind(sensor0, sensor1)
spplot(merged_sensors, "gpstime")
```
[image]

`lasrangecorrection()` is scrambling the sensor positions. They are good going in, then the function fails, and the sensor positions are bad: 
[image]

Jean-Romain, 2020-03-27:
Ok I think I got it. We have a side effect that mess up the original sensor shapefile. But internally in `lasrangecorrection()` it should not affect the processing. Please send be the two shapefile.

lucas-johnson, 2020-03-27:
[bad_sensor_positions.rds](https://github.com/Jean-Romain/lidR/files/4395317/bad_sensor_positions.txt)
[good_sensor_positions.rds](https://github.com/Jean-Romain/lidR/files/4395318/good_sensor_positions.txt)

Jean-Romain, 2020-03-27:
I fixed the mess put in the original dataset. I don't think it will fix the problem.

You don't need to merge the two tracks. They share the same gpstimes so only one measure will be retained anyway. Try with only one.

Try with the two tracks merged after updating the patch I pushed. If it fixes the issue I don't understand why. But who knows...

lucas-johnson, 2020-03-27:
It works now, thank you. With one set of tracks it works. The merged tracks work as well but with warnings.

Jean-Romain, 2020-03-27:
Yes, as I said almost all your gpstime are duplicated because you computed the aircraft positions twice. It retains only one value out of the two otherwise it can't interpolate between two values.

Jean-Romain, 2020-03-27:
I'm glad it works. If you make more tests in more dataset please tell me. The more I have feedback the more I'll trust the tool. Then I will be able to make the LAScatlog version.

lucas-johnson, 2020-03-27:
Sounds good, will do!

Jean-Romain, 2020-03-31:
Now the function `normalize_intensity()` formerly `lasrangecorrection()` works with a `LAScatalog`.

commit 51d9ff94 (2020-03-31): normalize_intensity with LAScatalog (close #336)
closed 2020-03-31 by Jean-Romain