r-lidar/lidR #339: [2.2.4] 'integer overflow' in lasreoffset

spono, 2020-04-30:
ciao JR,
I'm trying to shift a point cloud using `lasreoffset` but I get this error:

```
las = lasreoffset(las, xoffset = 1306977, yoffset = 5661354, zoffset=265.8018)

> Error: Incorrect xoffset: integer overflow.
```

May I ask you some more details about the kind of error? what does "integer overflow" mean?

The point cloud is a 3x3 metres scan derived from photogrammetry (i.e. local coordinates, with 0,0 in the centre of the point cloud) in which the scale has been fixed to 0.001 (using `lasrescale`).
The tool works fine with ALS point clouds so I guess the error is specific to the kind of dataset, but I don't understand how.

Thanks in advance

Jean-Romain, 2020-04-30:
Why do you want to reoffset? Are you sure to understand what that means? Please explain your intention so I can explain what you are doing wrong. Ideally a well formulated question should belong in  https://gis.stackexchange.com/ with the tag `lidr` so it may help other users as well.

spono, 2020-05-01:
done, you can delete this issue from here

closed 2020-05-01 by spono

Jean-Romain, 2020-05-01:
Move here https://gis.stackexchange.com/questions/360247/processing-photogrammetry-point-clouds-with-lidr