r-lidar/lidR #341: I have a mistake about writeLAS

LiuQW-YNnormal, 2020-05-03:
when I used lastrees, the las data can't wirten to a new file.
The code is 
**library(lidR)
las1 <- readLAS("D:/testpoint/project.las")
las <- lastrees(las1, li2012(R = 2,speed_up = 5))
plot(las,color='treeID')
writeLAS(las,"D:/testpoint/project1.las")**
or 
**library(lidR)
las1 <- readLAS("D:/testpoint/project.las",select="xyz")
writeLAS(las1,"D:/testpoint/project1.las")**
they all error 
** Invalid header: X scale factors should be factor ten of 0.1 or 0.5 or 0.25 not 9.8699e-09**

Jean-Romain, 2020-05-03:
You also have a similar warning at read time right ? See

- https://gis.stackexchange.com/questions/333324/rescaling-las-file-with-lidr-package-in-r
- https://gis.stackexchange.com/questions/360247/rescaling-and-reoffsetting-a-photogrammetry-point-cloud-with-lidr

closed 2020-05-03 by Jean-Romain

LiuQW-YNnormal, 2020-05-03:
Thanks for you very mach