r-lidar/lidR #343: plot_xyzrgb error "id must contain integers"

bi0m3trics, 2020-05-10:
Using `lidR `3.0.0 at the moment, but I saw this in the prior version and never reported it so I think this is the right repo (and not `lidR`)...
I can consistently reproduce the error with the following code and I seem to recall seeing it using objects returned by other functions besides `classify_ground` (or `lasground`)...
```
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile)
las <- classify_ground(las, csf()) #pmf() yields the same plotting error
plot(las, color="Classification", backend = "pcv")
```
Results in...
```
Error in plot_xyzrgb(x, y, z, col[1, ], col[2, ], col[3, ], id, size) : 
  'id' must contain integers
```

Jean-Romain, 2020-05-10:
Fixed. I came from `lidR`. Thank you for reporting. You must know that I never use `PointCloudViewer` myself. So if you find something weird or unexpected do not hesitate to report. I don't know if it works well and the development is frozen.

closed 2020-05-10 by Jean-Romain

bi0m3trics, 2020-05-10:
I love it, esp for viewing the clouds from our mobile scanner... it consistently performs well for ~30-50 million points and it allows me to do some quick diagnostics when `rgl` takes ages to display. 
Speaking of development, any plans to add any of the functionality you've added in the past few versions of `lidR`? Specifically, the `add_*` functionality is nice for adding to a point cloud scene but it clearly only works for `backend="rgl"`... If not (i'd imagine it's a ton of work), maybe mention it in the `lidR`'s documentation (i.e., `add_*` only works with `backend = "rgl"`).

Jean-Romain, 2020-05-10:
No plan because it is technically impossible. Once displayed everything is blocked until you close the windows. Thus:

```r
plot(las) %>% add_treetop3d(...)
```

Can reach the `add_treetop` statement only once the windows is close.  I would need to develop something that postpone the rendering but maintaining all backward compatibility. No way.

I add clarifications in the doc.

bi0m3trics, 2020-05-10:
That's what I figured... but it never hurts to ask. Thanks!

commit eb8d85c0 (2020-05-10): Fix #343