r-lidar/lidR #348: Crown segmentation algorithms do not work when run outside `segment_trees()` function

ptompalski, 2020-06-08:
When performing crown segmentation with either of the raster-based algorithms (e.g. silva2016, or dalponte2016), a following error message is returned:

`Error in if (!received_context %in% expected_contexts) stop(glue::glue("The '{name}' algorithm has not been called in the correct context."),  : 
  argument is of length zero`

In the code below everything runs ok except the last line:

```
library(lidR)
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile, select = "xyz", filter = "-drop_z_below 0")

chm = grid_canopy(las, 1, p2r())
ttops = find_trees(chm, lmf(3))
crowns = dalponte2016(chm, ttops)()
```

Jean-Romain, 2020-06-08:
It works for me. I guess you are using a development version that has been updated at a very bad moment when I introduced this bug. But I think I fixed that recently before to release 3.0.0. Try to update first from `master` that know contains the released version 3.0.0

ptompalski, 2020-06-08:
yup, works fine after updating.

closed 2020-06-08 by ptompalski