r-lidar/lidR #350: Grid_Terrain In absence of class 2

nidhi23ug, 2020-06-11:
First just to mention, I have used liDR extensively all through my doctoral thesis and I can not thank you enough for such great and easy tools. 

There has always been one issue that I wanted to raise is the issue of Grid_Terrain when there is no class 2. So one data provider did not follow LAS specifications and classified ground points in class 1 and 2 as unclassified points, and in some tiles, there are no unclassified points. So obviously in some tiles, there is no 2. 

However, when I use class 1L for grid_terrain, for some tiles where 2L is missing I keep on getting _"No ground points found. Impossible to compute a DTM."_ I have checked the source code and I think this is forced in even if class chosen for interpolation is different. 

I know how to surpass it, I just kept Null class as 2 and I can still run it. But I wish I do not have to do it.

nidhi23ug, 2020-06-11:
`lidDat = readLAS(Path)
> unique(lidDat@data$Classification)
[1] 6 1 8
> dtm= lidR::grid_terrain(lidDat,res = 0.25,algorithm=knnidw(k=10L),use_class = c(1L))
Error: No ground points found. Impossible to compute a DTM.`

Jean-Romain, 2020-06-11:
Try to use `-change_classification_from_to 1 2` in `filter`. This will reclassify at read time

```r
readLAS(path, filter = "-change_classification_from_to 1 2")
```

Jean-Romain, 2020-06-11:
Also 

```r
grid_terrain(lidDat,res = 0.25,algorithm=knnidw(k=10L),use_class = c(1L, 2L))
```

Should do the job no?

nidhi23ug, 2020-06-11:
> ```r
> use_class = c(1L, 2L))
> ```

No, this one does not do, as there is no 2 class in some. Yes, reclassification does help. I already tried doing the reclassification and it worked fine.

Jean-Romain, 2020-06-11:
If there is no class 2 then only class 1 will be used. I don't understand why `use_class = 1:2` cannot do the job.

nidhi23ug, 2020-06-11:
> If there is no class 2 then only class 1 will be used. I don't understand why `use_class = 1:2` cannot do the job.

Yes, this is precisely why I raised the issue. I could always reclassify but I didn't understand why it does not work. 

`> lidDat = readLAS(Path)
> unique(lidDat@data$Classification)
[1] 6 1 8
> dtm= lidR::grid_terrain(lidDat,res = 0.25,algorithm=knnidw(k=10L),use_class = c(1L, 2L))
Error: No ground points found. Impossible to compute a DTM.`

Jean-Romain, 2020-06-11:
Ho ok I got it. Will see that tomorrow. It look like a bug. Are you able to share a small file that I'll be able to use to reproduce.

nidhi23ug, 2020-06-11:
> Ho ok I got it. Will see that tomorrow. It looks like a bug. Are you able to share a small file that I'll be able to use to reproduce.

Yes sure, can I directly send it to you without keeping it public?

Jean-Romain, 2020-06-11:
As you wish

Jean-Romain, 2020-06-11:
Fixed. You can install from github or way few more day I will release v3.0.1 soon.

The problem was that when I introduced `use_class` options I didn't remove a test consting in counting the point classified 2 to fail early in absence of point. This invalidated your specific use case.

Thank you for reporting

closed 2020-06-11 by Jean-Romain

nidhi23ug, 2020-06-11:
> Fixed. You can install from github or wait few more day I will release v3.0.1 soon.
> 
> The problem was that when I introduced `use_class` options I didn't remove a test consting in counting the point classified 2 to fail early in absence of point. This invalidated your specific use case.
> 
> Thank you for reporting

Ah thank you so much for your quick action. Will do. Cheers

khaz666666, 2022-03-04:
Hi,
I have the same issue and the following steps did not help. I used QGIS (LAStools plugins) and I managed to get some reliable results. Since I have a lot of LAS files and I need to write a for loop, I cannot rely on QGIS and I need to get the Lidr package work. So, your help is highly appreciated!

Jean-Romain, 2022-03-04:
@khaz666666  You can't have the same issue I fixed this one.If you have questions on how to use lidR please go to https://gis.stackexchange.com/questions/tagged/lidr. It you found a bug please open a new issues. That on is close for 1 year and a half

commit 256fcfbd (2020-06-11): Fix #350