r-lidar/lidR #351: lidR grid_metrics netCDF support with layer names intact

lucas-johnson, 2020-06-12:
### Feature Request
I know that `grid_metrics()` can write to different formats by specifying the format as per this post: https://gis.stackexchange.com/questions/347064/keeping-name-of-metrics-as-band-names-when-using-grid-metrics-lidr

In following that post I found that the only way to store layer names through writeRaster is to use the `.grd` and `.gri` formats which necessitates 2 files. 

netCDF supports named layers in one file, though I think this requires avoiding the use of `writeRaster` and rather using the netCDF analog, so I am requesting this as an enhancement to the current offering. 

The current workaround is to have `grid_metrics()` write to netCDF, read the netCDF file back in, rename the the layers, and save the file again. I know it's a bit of a tiny detail, probably not a high priority, and more of a nicety than anything...

Jean-Romain, 2020-06-12:
Hi Lucas,

Layer names is indeed an annoying limitation for which I have concerns. I chose the tif format as default because I believed it is the most known format independently of the user's background in spatial processing (I mean beginners can use it easily).

Is layer names storage a limitation of the tif format itself or a limitation of `writeRaster` from `raster`? According to this issue https://github.com/rspatial/raster/issues/88 I'd think the later is correct.

Also I don't know anything about netCDF format. If you want to convince me about the relevance of your suggestion you must explain me how and where it will bring value to the package. But if it needs a supplementary package dependency to work I can already tell you that I cannot address your request. My current plan at mid-terms is to reduce the number of dependencies.

lucas-johnson, 2020-06-12:
Thanks for your response. I think you're right, it's not an issue with `lidR` but rather an issue with `raster` and/or `tif`. And yes, this request requires a supplementary package `ncdf4` (https://cran.r-project.org/web/packages/ncdf4/ncdf4.pdf) so I guess that means it's not possible to add. The benefit I am after is to simply store the names of each layer with the data in one file. But of course there are work arounds.

Jean-Romain, 2020-06-12:
You can add your own drivers to the catalog. So if you want it can write netCDF files. See also https://github.com/Jean-Romain/lidR/wiki/Modify-the-LAScatalog-drivers

closed 2020-06-12 by Jean-Romain