r-lidar/lidR #352: Small readme update

tiagodc, 2020-06-22:
Hey Jean, `libv8` is deprecated starting on Ubuntu 19. On Ubuntu 20.04 I had to install [`libnode-dev`](https://packages.ubuntu.com/search?keywords=libv8) to meet _lidR_'s dependencies. Just telling as you might consider mentioning it on the project's README.

Cheers

Jean-Romain, 2020-06-22:
Thank you for reporting that. I'll wait and see if I can experience the same. How do you know it is deprecated? I mean, I've never seen `apt-get` saying that a package is deprecated.

tiagodc, 2020-06-24:
Maybe I shouldn't have said _deprecated_, sorry about that. I just had a normal apt error saying that it couldn't find `libv8`, so I just checked the Ubuntu packages and saw that the `v8` libs are included under `libnode` on the more recent releases.

closed 2020-09-02 by Jean-Romain