r-lidar/lidR #353: `sp` being deprecated in favor of `sf`

mikoontz, 2020-06-22:
From Edzer Pebesma: [https://twitter.com/edzerpebesma/status/1272928917700165633](https://twitter.com/edzerpebesma/status/1272928917700165633).

> Yes, please consider sp deprecated. For any new work use sf, and sf only. In case there are real speed issues, please report an sf issue. If you have massive point data sets, consider using a single MULTIPOINT, or an entirely different solution.

Something to consider, as some of the `lidR` machinery relies on `sp`. I may be able to help migrate for `lidR` v3.0.0 if you like.

Jean-Romain, 2020-06-22:
Short answer. It's complicated. 

When I started lidR `sf` was in development. I could have moved to `sf` in v2.0.0 but `sf` have many performance issues. I may be wrong but I'd say that `sf` is for geoprocessing and `sp` is for programming. It is the feeling I have even if I don't know very well `sf`. Now it is almost impossible to move. It is not only some internal machinery that relies on `sp` it is the whole foundations. And what does not rely on `sp` relies on `raster` which rely on `sp` and does not support `sf`. 

So it is complete re-engineering from 0 that is required not only some tweaks here and there to change the input/output from `sp` to `sf`.

mikoontz, 2020-06-23:
Yeah, fair enough! I imagine `sp` and `raster` might be soft deprecated for a while, so perhaps this ought to be a longer-term goal for `lidR` v4.0.0 rather than v3.0.0?

I think `terra` is being written to succeed `raster` and perhaps `stars` or `exactextractr` might be useful when it comes time to eliminate dependency on `raster`.

Regardless, it seems like some of the key infrastructure of `lidR` will eventually (but sooner rather than later) be succeeded by more modern spatial tools in `R`, and because `lidR` is the best las manipulator in `R`, I'd be happy to try to help with some conversions when the time is right :)

Jean-Romain, 2020-06-23:
I'm opened to suggestion.

closed 2020-08-18 by Jean-Romain

wiesehahn, 2021-10-15:
Any new thoughts about the implementation of `sf` and `terra`?

Jean-Romain, 2021-10-15:
@wiesehahn  I'm working hard on lidR 4.0.0 that will no longer depend on `sp/raster` and only `sf/stars`. The challenge is being 99% backward compatible. See https://github.com/Jean-Romain/lidR/discussions/471