r-lidar/lidR #361: No add_treetops3d plot window

cyril06, 2020-07-22:
Hello,
When a tree top object is cropped by an extent it is impossible to plot with add_treetops3d.

ttops <- tree_detection(ctg, lmf(5))
l<-readLAS(ctg[50,])
temp<-crop(ttops,extent(l))
x<-plot(l)
add_treetops3d(x, temp)

The plot window doesn't appear.

Am I doing something wrong ?

Regards

Jean-Romain, 2020-07-22:
This is a little short to give you an answer. The code looks good. Are you sure that the window does not appear because the rendering is too long. How many points are they? what is the size of the tile? what is the point density? Have you tried with a smaller example? are you able to produce a reproducible example? Please provide details on your issue.

cyril06, 2020-07-22:
When you launch the command:
add_treetops3d(x, ttops)
Everything is fine.

What is the real difference between ttops and temp ? Maybe there is a bug somewhere.

Regards

Jean-Romain, 2020-07-22:
As I said without a reproducible example I can hardly help you. The following works:

```r
LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las <- readLAS(LASfile)
temp <- find_trees(las, lmf(ws = 5))

x <= plot(las)
add_treetops3d(x, temp)
```

cyril06, 2020-07-22:
Maybe there is something wrong like not enough memory. I'll try later when I have restarted my R session.

Now when I launch:
x=plot(l)
The plot window appears.

Regards

closed 2020-07-23 by Jean-Romain