r-lidar/lidR #362: plot_crossection

stefanoch90, 2020-07-24:
Hi, i have tried to use the command: plot_crossection but R tell me "I can't find the function "plot_crossection".

Jean-Romain, 2020-07-24:
If you have a question about the book please ask on the [book repo](https://github.com/Jean-Romain/lidRbook). Thanks. The function is defined [here](https://jean-romain.github.io/lidRbook/io.html#plot-crossection)

closed 2020-07-24 by Jean-Romain

williamlai2, 2022-09-02:
Hi, I am just following up on the `plot_crossection()` function. 

> The function is defined [here](https://jean-romain.github.io/lidRbook/io.html#plot-crossection)

That link just leads to a 404 and it doesn't seem to be defined anywhere else in the book.

kadyb, 2022-09-04:
@williamlai2, the book is now available in this repository: https://github.com/r-lidar/lidRbook. This is the corrected link to `plot_crossection()` function: https://r-lidar.github.io/lidRbook/io.html#plot-crossection

kathno, 2023-10-24:
Hi, I have the same problem "could not find function "plot_crossection" - the link and the book do not really help. Even if I run the example script with the example data, it could not find the function. 
Did you solve xour problem? What was the mistake?

jameslamping, 2023-10-24:
That function is not part of the R package, but it is included in the example book in GitHub. I use it all the time because it's super helpful in looking at how well ground classification worked! Here is the function as presented:

```

plot_crossection <- function(las,
                             p1 = c(min(las@data$X), mean(las@data$Y)),
                             p2 = c(max(las@data$X), mean(las@data$Y)),
                             width = 4, colour_by = NULL)
{
  colour_by <- rlang::enquo(colour_by)
  data_clip <- clip_transect(las, p1, p2, width)
  p <- ggplot(data_clip@data, aes(X,Z)) + geom_point(size = 0.5) + coord_equal() + theme_minimal()

  if (!is.null(colour_by))
    p <- p + aes(color = !!colour_by) + labs(color = "")

  return(p)
}

```