r-lidar/lidR #364: install error

fipoucat, 2020-08-03:
I am attempting to install from github lidR on mac with R 4.0.2 but always get error:
ERROR: loading failed
* removing ‘/Library/Frameworks/R.framework/Versions/4.0/Resources/library/lidR’
* restoring previous ‘/Library/Frameworks/R.framework/Versions/4.0/Resources/library/lidR’
Error: Failed to install 'lidR' from GitHub:
  (converted from warning) installation of package ‘/var/folders/wb/pry3ghxn4s30_yrgyynbrxs80000gn/T//RtmplMlzPy/fileac2d6c7c4add/lidR_3.0.3.tar.gz’ had non-zero exit status

Any hint on a possible solution to fix this?

Best regards

fipoucat, 2020-08-03:
below the full output:
> devtools::install_github("Jean-Romain/PointCloudViewer")
Downloading GitHub repo Jean-Romain/PointCloudViewer@HEAD
✓  checking for file ‘/private/var/folders/wb/pry3ghxn4s30_yrgyynbrxs80000gn/T/RtmplMlzPy/remotesac2d288888f8/Jean-Romain-PointCloudViewer-6101481/DESCRIPTION’ ...
─  preparing ‘PointCloudViewer’:
✓  checking DESCRIPTION meta-information ...
─  cleaning src
─  checking for LF line-endings in source and make files and shell scripts
─  checking for empty or unneeded directories
─  building ‘PointCloudViewer_0.2.0.tar.gz’
   
* installing *source* package ‘PointCloudViewer’ ...
** using staged installation
** libs
clang++ -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I/usr/local/include   -fPIC  -Wall -g -O2  -c RcppExports.cpp -o RcppExports.o
clang++ -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I/usr/local/include   -fPIC  -Wall -g -O2  -c camera.cpp -o camera.o
In file included from camera.cpp:1:
./camera.h:4:10: fatal error: 'SDL/SDL.h' file not found
#include <SDL/SDL.h>
         ^~~~~~~~~~~
1 error generated.
make: *** [camera.o] Error 1
ERROR: compilation failed for package ‘PointCloudViewer’
* removing ‘/Library/Frameworks/R.framework/Versions/4.0/Resources/library/PointCloudViewer’
Error: Failed to install 'PointCloudViewer' from GitHub:
  (converted from warning) installation of package ‘/var/folders/wb/pry3ghxn4s30_yrgyynbrxs80000gn/T//RtmplMlzPy/fileac2d1bd86692/PointCloudViewer_0.2.0.tar.gz’ had non-zero exit status

Jean-Romain, 2020-08-03:
Are you talking about `lidR 3.0.3` or `PoinCloudViewer`?

fipoucat, 2020-08-03:
Sorry for that, I am talking about lidR but then did a try on PoinCloudVieewer, correct output below:

devtools::install_github("Jean-Romain/lidR")
Downloading GitHub repo Jean-Romain/lidR@HEAD
✓  checking for file ‘/private/var/folders/wb/pry3ghxn4s30_yrgyynbrxs80000gn/T/RtmplMlzPy/remotesac2d6f3fff78/Jean-Romain-lidR-c08e9fe/DESCRIPTION’ (352ms)
─  preparing ‘lidR’:
✓  checking DESCRIPTION meta-information ...
─  cleaning src
─  checking for LF line-endings in source and make files and shell scripts
─  checking for empty or unneeded directories
─  building ‘lidR_3.0.3.tar.gz’
   
* installing *source* package ‘lidR’ ...
** using staged installation
** libs
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c BoundingBox.cpp -o BoundingBox.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c C_Gatziolis2019.cpp -o C_Gatziolis2019.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c C_dalponte2016.cpp -o C_dalponte2016.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c C_delaunay.cpp -o C_delaunay.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c GridPartition.cpp -o GridPartition.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c LAS.cpp -o LAS.o
LAS.cpp:268:10: warning: unused variable 'i' [-Wunused-variable]
  double i;
         ^
LAS.cpp:569:11: warning: unused variable 'p' [-Wunused-variable]
    Point p;
          ^
LAS.cpp:601:11: warning: unused variable 'p' [-Wunused-variable]
    Point p;
          ^
3 warnings generated.
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c Progress.cpp -o Progress.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c RcppExports.cpp -o RcppExports.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/RcppArmadillo/include' -I/usr/local/include   -I../inst/include/ -DRCPP_NO_MODULES -fPIC  -Wall -g -O2  -c RcppFunction.cpp -o RcppFunction.o
clang++ -mmacosx-version-min=10.13 -std=gnu++11 -I"/Library/Frameworks/R.framework/Resources/include" -DNDEBUG  -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/BH/include' -I'/Library/Frameworks/R.framework/Versions/4.0/Resources/library/Rcpp/in [... cut ...]
 in /usr/local/lib/libquadmath.0.dylib
Error: loading failed
Execution halted
ERROR: loading failed
* removing ‘/Library/Frameworks/R.framework/Versions/4.0/Resources/library/lidR’
* restoring previous ‘/Library/Frameworks/R.framework/Versions/4.0/Resources/library/lidR’
Error: Failed to install 'lidR' from GitHub:
  (converted from warning) installation of package ‘/var/folders/wb/pry3ghxn4s30_yrgyynbrxs80000gn/T//RtmplMlzPy/fileac2d529d5857/lidR_3.0.3.tar.gz’ had non-zero exit status

Best regards

Jean-Romain, 2020-08-03:
Sorry I don't know. There some missing function definition somewhere but not in `lidR`. The interesting part is

```r
Error: package or namespace load failed for ‘lidR’ in dyn.load(file, DLLpath = DLLpath, ...):
 unable to load shared object '/Library/Frameworks/R.framework/Versions/4.0/Resources/library/00LOCK-lidR/00new/lidR/libs/lidR.so':
  dlopen(/Library/Frameworks/R.framework/Versions/4.0/Resources/library/00LOCK-lidR/00new/lidR/libs/lidR.so, 6): Symbol not found: ___addtf3
  Referenced from: /usr/local/lib/libquadmath.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib
 in /usr/local/lib/libquadmath.0.dylib
```
But you are on MacOS so I can't help you. Ask on internet. Or install `lidR` from CRAN. There is no good reason to install the package from github instead of CRAN.

closed 2020-08-03 by Jean-Romain