r-lidar/lidR #366: Question: notes for compiling lidR on HPC or machine with openMPI

serbinsh, 2020-08-12:
Hello

I see in the docs that you can setup lidR to leverage multi-core or HPC machines to use multiple threads/cores but it wasnt clear how it is suggested to compile the pacakge to leverage this?  I saw this on the wiki: https://github.com/Jean-Romain/lidR/wiki/Make-cheap-High-Performance-Computing-to-process-large-datasets

While pretty cool it wasnt exactly what I was looking for.  Would I just need to compile from source using mpicc instead of gcc?  Sorry for the silly question, I just wasnt sure what the correct approach would be

Thanks!

serbinsh, 2020-08-12:
OK as I suspected, the trick is to make the mpi libs availible when compiling from source.  On my linux cluster I loaded my openMPI module and when it compiled I could note the "-fopenmp" flag

when i load the library I see

```
> library('lidR')
Loading required package: raster
Loading required package: sp
lidR 3.0.3 using 8 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/Jean-Romain/lidR>.
```

and I can change this with

```
set_lidr_threads(16)
```

and see with

```
> get_lidr_threads()
[1] 16
```

So I just need to recompile on my Mac and tell R to not use clang but my gcc with openMPI support

Jean-Romain, 2020-08-12:
There is nothing to do to leverage OpenMP (where available). The compilation options are already the good one on Linux. On Windows the CRAN build binaries with OpenMP support. The only issue is on MacOS where the CRAN binaries are not built with OpenMP support. But I can't help you on Mac. If you encounter troubles you'd better to ask somewhere else e.g. on stackoverflow.

Jean-Romain, 2020-08-12:
See also 
```
?lidR::`lidR-parallelism`
```

closed 2020-08-12 by Jean-Romain

serbinsh, 2020-08-12:
Thanks.  Sorry for the hassle.  Yes it was much easier on my linux HPC

However, to close the loop on how I was successful on Mac

I setup my ~/.R/Makevars as such

```VER=-9
FC=gfortran$(VER)
CC = /usr/local/Cellar/llvm/10.0.0_3/bin/clang
CXX = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp -std=gnu++11
CXX98 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX11 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX14 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX17 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp

FLIBS=-L/usr/local/Cellar/gcc@9/9.3.0/lib/gcc/9 -lgfortran -lquadmath -lm
```

Then compiled from source

install.packages('lidR_3.0.3.tar.gz', repo=NULL, type='source')

Then open the library and 

```
Loading required package: raster
Loading required package: sp
lidR 3.0.3 using 6 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/Jean-Romain/lidR>.
```

So its possible

serbinsh, 2020-08-12:
And I can confirm its using MPI

```> ctg <- readLAScatalog("~/Data/RS_GIS_Data/UAS/20180726_Kougarok_Flight6/Level_1/NGEEArctic_Seward_Osprey_Kougarok_20180726_Flight6_CloudPoints.laz")
> ctg
class       : LAScatalog (v1.2 format 2)
extent      : 508160.5, 508412.7, 7226676, 7226848 (xmin, xmax, ymin, ymax)
coord. ref. : +proj=utm +zone=3 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0
area        : 43465.88 m²
points      : 80.35 million points
density     : 1848.6 points/m²
num. files  : 1
> set_lidr_threads(6)
> out <- tree_detection(ctg, lmf(2))
```

[image]

[image]

Jean-Romain, 2020-08-12:
Good to know. Can you write down the exact set-up required on Mac to get OpenMP support so I can point to this issue if somebody else ask me the question. Thanks

serbinsh, 2020-08-17:
@Jean-Romain Sure, here is a summary:

1) brew install llvm 
-- (to replace default mac clang/clang++ with llvm version)
2) brew install openmpi
3) Create or update your R Makevars file with compiler info

Mine looks like this:

```
FC=gfortran$(VER)
CC = /usr/local/Cellar/llvm/10.0.0_3/bin/clang
CXX = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp -std=gnu++11
CXX98 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX11 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX14 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp
CXX17 = /usr/local/Cellar/llvm/10.0.0_3/bin/clang++ -fopenmp

FLIBS=-L/usr/local/Cellar/gcc@9/9.3.0/lib/gcc/9 -lgfortran -lquadmath -lm
```

NOTE: I had to add openMP flags ("-fopenmp") to get lidR to compile and utilize openMPI

4) Compile from source ```install.packages('lidR_3.0.3.tar.gz', repo=NULL, type='source')```
-- because you updated your ~/.R/Makevars R will override its default compiler options with those listed in that file.  NOTE: This will change the compiler options R-wide so if you dont want to use these with all R lib installs, then you would need to comment out these options once you build lidR....its a bit clunky.  Or you can just use these settings and any package source will be updated with these newer options.

serbinsh, 2021-12-09:
For completeness I wanted to say that you dont need to follow my instructions using local source, you can also successfully build a parallel implementation on mac OS like this

install.packages('lidR', type='source')

After making sure to update/add and edit the .R/Makevars like above, but with updated info based on which version of the compilers you are using

ColinPSweeney, 2024-06-14:
Following up on @serbinsh steps from a few years back. I am trying to get OpenMP set up on a Mac (Sonoma 14.3.1) for lidR. 

I've been successful at steps 1 and 2 as written. For step 3 I created a new "~/.r/Makevars.txt" updated the filepaths as best I can:

FC=gfortran$(VER)
CC = /usr/local/Cellar/llvm/18.1.7/bin/clang
CXX = /usr/local/Cellar/llvm/18.1.7/bin/clang++ -fopenmp -std=gnu++11
CXX98 = /usr/local/Cellar/llvm/18.1.7/bin/clang++ -fopenmp
CXX11 = /usr/local/Cellar/llvm/18.1.7/bin/clang++ -fopenmp
CXX14 = /usr/local/Cellar/llvm/18.1.7/bin/clang++ -fopenmp
CXX17 = /usr/local/Cellar/llvm/18.1.7/bin/clang++ -fopenmp

FLIBS=-L/usr/local/Cellar/gcc/14.1.0_1/lib/gcc/14 -lgfortran -lquadmath -lm

However for step 4, I ran install.packages('lidR', type='source') but got a non-zero exit status. Any chance the compilers/file paths need to be adjusted as I've written them?

Also, on other OpenMP pages, I was seeing others recommend libomp over llvm. Any thoughts on this?
(https://github.com/Rdatatable/data.table/wiki/Installation )