r-lidar/lidR #368: open pattern parameter of list.files for readLAScatalog and add more detailed documentation

Lenostatos, 2020-08-15:
Dear lidR developers,

I was using the `readLAScatalog` function to read in a bunch of LAZ files. Because my directory contained two sets of files, one ending in "CIR.laz" and the other in "RGB.laz", I wanted to use the `pattern` parameter of the `list.files` function to only use one of the two sets, i.e. I would have preferred to write:

`readLAScatalog(folder = "path/to/my/folder/", pattern = "*CIR\\.laz")`

However, this results in the error _"formal argument "pattern" matched by multiple actual arguments"_. As far as I understand it, this is because `readLAScatalog` already passes an argument to the `pattern` parameter internally. Therefore my code looks like this right now:

`readLAScatalog(list.files("path/to/my/folder/", pattern = "*CIR\\.laz", full.names = TRUE))`

This is of course not a big issue. I was just excited to see the hint in the documentation for the `...` parameter about "Extra parameters to list.files". For me, this description seemed to mean "You don't have to use list.files explicitly here. Just pass any parameters that you would otherwise pass to list.files.".

**TLDR:**
I would be happy if `readLAScatalog` allowed its users to pass an argument to the `pattern` parameter of `list.files`, without having to call `list.files` explicitly. Maybe this is too much of a breaking change but in any case I think the documentation of the ... parameter to `readLAScatalog` should be extended a bit. Maybe something like
_"Extra parameters to list.files. Typically `recursive = TRUE`. Please note that the pattern parameter of list.files is not supported."_

What are your thoughts on this?

Jean-Romain, 2020-08-15:
`pattern` is indeed hard coded to retain only las/laz file and providing `pattern` in `...` is equal to pass `pattern` twice. If fixed it. Thank you for reporting

closed 2020-08-15 by Jean-Romain

Lenostatos, 2020-08-15:
Thank you for the quick fix!