r-lidar/lidR #375: voxel_metrics with "empty" voxels?

jgrn307, 2020-09-12:
Is it possible to run voxel_metrics using the "bounding cube", so all "empty" voxels also are included in the output?

Jean-Romain, 2020-09-12:
No, but it is something that was asked many times (not always explicitly like that) so I guess this is something that is needed by several persons and it worth a feature request. However, waiting for a native implementation, it is easy to add missing voxels in the output and fill the metrics with NAs.

Jean-Romain, 2020-09-28:
New argument `all_voxels = TRUE` in `voxel_metrics()`

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile)

vm = voxel_metrics(las, ~list(Imean = mean(Intensity)), 3)
avm = voxel_metrics(las, ~list(Imean = mean(Intensity)), 3, all_voxels = T)

nrow(vm)
#> 23708
nrow(avm)
#> 66913

plot(vm, color = "Imean", colorPalette = heat.colors(50), trim = 90)
plot(avm, color = "Imean", colorPalette = heat.colors(50), trim = 90)
```

[image]

closed 2020-09-28 by Jean-Romain