r-lidar/lidR #377: add_lasattribute for merge_spatial?

Lenostatos, 2020-09-19:
Dear lidR developers,

when using `merge_spatial` it would be nice if I had an option `add_lasattribute` like the one of `normalize_height` that allows to save the merged spatial data when writing the `LAS` object to a file. Of course I can always call the function `add_lasattribute` explicitly after calling `merge_spatial` so this is just about a convenience feature.

Thank you for your consideration!
Leon

Jean-Romain, 2020-09-19:
I'll think about it. But I don't think I'll do it. 

The first difference between `normalize_height` and `merge_spatial` is that in the first one we know that the data type is decimal number but for the later it can be any R type including factors, character, boolean and so on. Most of these types cannot be written in LAS files. Thus I think that most use cases correspond to invalid attribute addition leading to warning or errors. I prefer the user to get an error when doing `add_lasattribute` instead of dealing internally with all the possible cases to protect error from `add_lasttribute` in `merge_spatial`. 

The second difference is that `normalize_height` we know the resolution of the coordinate which allows for storage optimization with `add_lasattribute_manual`. In `merge_spatial` if an optimization is possible it cannot be guessed internally. Only the users who have knowledge on their data can chose to use `add_lasattribute` or `las_addattribute_manual`

Jean-Romain, 2020-09-19:
I can also say that with a RGB stack it is meaningless to add an attribute because 3 cores attributes are added. So I'm closing because I cannot retain you suggestion.

closed 2020-09-19 by Jean-Romain