r-lidar/lidR #378: Implementation of Graph Based Segmentation (Strimbu V and Strimbu B, 2015)

azh3, 2020-09-26:
I am working w/ Dr. Strimbu to implement graph-based segmentation in R and was curious if it could be implemented within the LidR package alongside other segmentation algorithms. The procedure is computationally complex and has been implemented in Java, I have the code, but haven't worked in Java before and don't know where I'd begin in translating it or if it would be better to build something from scratch. The paper is: 

Strimbu V and Strimbu B (2015) A graph-based segmentation algorithm for tree crown extraction using airborne LiDAR data. ISPRS Journal of Photogrammetry and Remote Sensing 104, 30-43 DOI: [10.1016/j.isprsjprs.2015.01.018](https://www.sciencedirect.com/science/article/pii/S0924271615000544?via%3Dihub)

According to the authors: "This work proposes a segmentation method that isolates individual tree crowns using airborne LiDAR data. The proposed approach captures the topological structure of the forest in hierarchical data structures, quantifies topological relationships of tree crown components in a weighted graph, and finally partitions the graph to separate individual tree crowns. This novel bottom-up segmentation strategy is based on several quantifiable cohesion criteria that act as a measure of belief on weather two crown components belong to the same tree. An added flexibility is provided by a set of weights that balance the contribution of each criterion, thus effectively allowing the algorithm to adjust to different forest structures"(Strimbu V and Strimbu B, 2015).

The results of this segmentation routine were very accurate compared to other methods, with an accuracy index of 74.75% for hard/complex/heterogeneous stands, 92.25% for medium complexity stands, and 98.98% for low complexity (park/plantation) stands. The algorithm also allows and corrects for multiple local maxima, which is key when delineating crowns that may have multiple "treetops."

I would appreciate any assistance in the implementation of these algorithms in R, hopefully within the lidR package! My hope is to implement this algorithm in R and use it for urban tree crown 

- [ ] delineation.

[image]

Jean-Romain, 2020-09-26:
Hi,

I already read the paper and to be honest I did not understood everything, many parts of the procedure relying on methods I'm not familiar with.

I'm really keen to help you implementing this method in lidR from the original source code and I already did it for other algorithm such as [csf](https://github.com/Jean-Romain/RCSF), [mcc](https://github.com/Jean-Romain/RMCC), [Gatzioli2019](https://github.com/Jean-Romain/lidR/blob/master/src/C_Gatziolis2019.cpp). However in your case we are starting from Java and I know virtually nothing about Java. So I won't help you much and you will have to get your hands dirty.

I can easily guess that the code is complex. So it probably won't go within lidR and should instead belong in it's own package like [csf](https://github.com/Jean-Romain/RCSF) and [mcc](https://github.com/Jean-Romain/RMCC). *Maybe* you can directly interface the java code with R using [`rJava`](https://cran.r-project.org/web/packages/rJava/index.html).  As I said I'm not knowledgeable about Java so I won't be of great help here. I'm keen to help you but you will have to make the most of the job yourself.

azh3, 2020-09-26:
Hi, thanks for the tip! 

I don't have any familiarity with Java or graph theory (yet), but I'll keep you posted, still in the very early stages of the project.

closed 2020-09-27 by Jean-Romain