r-lidar/lidR #383: *Error: cannot allocate vector of size ...**

azh3, 2020-10-17:
I'm getting this error when I attempt to run a function using a 6.5 Gb LiDAR dataset:

plan(multisession, workers = 32L)
set_lidr_threads(32L)

DTM <- lidR::grid_terrain(las = LidRtst, res = 1, algorithm = lidR::tin(), use_class = c(2L, 8L, 9L))

traceback()
3: rOverlay(las, res)
2: grid_terrain.LAS(las = LidRtst, res = 1, algorithm = lidR::tin(), 
       use_class = c(2L, 8L, 9L))
1: lidR::grid_terrain(las = LidRtst, res = 1, algorithm = lidR::tin(), 
       use_class = c(2L, 8L, 9L))

**Error: cannot allocate vector of size 4211100.6 Gb**

**LiDAR Info:** 
Class: | LAS (v1.4 format 6)
Memory: | 6.8 Gb
Extent: | -21457120, 21163158, -21259604, 21133562 (xmin, xmax, ymin, ymax)
Coord. Ref.: NAD83(2011) / Oregon North (Intl ft)
Area: | 1,733,073,556 kunits²
Points: | 76.32 million points
Density: | 17.709 points/units²

**Session Info:** 
R version 4.0.3 (2020-10-10)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server x64 (build 17763)

**System Info**
OS: Microsoft Windows Server 2019 Standard Version 10.0.17763
Processors: 2 x Intel(R) Xeon(R) CPU E5-2620 0 @ 2.00 GHz, 1995 Mhz, 6 Cores(s), 12 Logical Processor(s)
RAM: 128 GB

Jean-Romain, 2020-10-17:
The extent of your point cloud is

```
-21457120, 21163158, -21259604, 21133562
```

which correspond to 1 trillion 733 billion and few more millions square feet. This is the size of the expected raster. I don't know what is wrong with your dataset but you should start looking if your data are correct.

azh3, 2020-10-17:
Aha! Thanks for pointing that out! I will look into that.

closed 2020-10-17 by Jean-Romain