r-lidar/lidR #386: Error in Processing Chunk las Catalog

azh3, 2020-10-26:
I keep getting this error and have tried everything I can think of to figure it out, any idea what the issue might be?

**Error in if (ext %in% c("las", "laz")) any_type = "las" else if (ext %in% : argument is of length zero**

**callback:**
4: catalog_merge_results(ctg, output, mer, as.character(substitute(FUN)))
3: catalog_apply(las, grid_canopy, res = res, algorithm = algorithm, 
       .options = options)
2: grid_canopy.LAScatalog(las = LidRtstn, res = 1, algorithm = lidR::pitfree(thresholds = c(6.56168, 
       ht), c(9.84252, 6.56168), subcircle = 0.66))
1: lidR::grid_canopy(las = LidRtstn, res = 1, algorithm = lidR::pitfree(thresholds = c(6.56168, 
       ht), c(9.84252, 6.56168), subcircle = 0.66))

**LAS Summary:**
class       : LAScatalog (v1.4 format 6)
extent      : 7648120, 7653883, 675298.5, 681872.1 (xmin, xmax, ymin, ymax)
coord. ref. : NAD83(2011) / Oregon North (ft) 
area        : 36.17 kft²
points      : 34.11 million points
density     : 0.9 points/ft²
num. files  : 12 
proc. opt.  : buffer: 100 | chunk: 0
input opt.  : select: * | filter: -keep_class 2 3 4 5 8 9
output opt. : on disk | w2w guaranteed | merging enabled
drivers     :
 - Raster : format = GTiff  NAflag = -999999  
 - LAS : no parameter
 - Spatial : overwrite = FALSE  
 - SimpleFeature : quiet = TRUE  
 - DataFrame : no parameter

R version 4.0.2 (2020-06-22)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server x64 (build 17763), RStudio 1.3.1093
lidR v3.0.4

[image]

azh3, 2020-10-26:
The issue appears to have been resolved after reading the normalized LAS catalog separately, opposed to assigning the output of the normalize_height function, I also indexed the normalized .las files which doesn't appear to be carried out by the normalize_height function on its own and could affect the speed of any subsequent processing. 

lidR::opt_output_files(LidRtst) <- paste0(outdir, "/NORMLAS/{ID}_NORM")
lidR::normalize_height(las = LidRtst, algorithm = lidR::tin(), na.rm = TRUE, use_class = c(2L, 8L, 9L))
LidRtstn <- lidR::readLAScatalog(folder = paste0(outdir, "/NORMLAS"))
lidR::opt_wall_to_wall(LidRtstn) == TRUE
lidR::opt_chunk_buffer(LidRtstn) <- 100
lidR:::catalog_laxindex(LidRtstn)

opposed to:
Lidrtstn <- lidR::normalize_height(las = LidRtst, algorithm = lidR::tin(), na.rm = TRUE, use_class = c(2L, 8L, 9L))

I also edited the code chunk as below:
 
lidR::grid_canopy(las = LidRtstn, res = 1, algorithm = pitfree(thresholds = c(6.56168, 16.4042, 32.8084, 49.2126, 65.6168, 82.0210, 98.4252, 114.8294), max_edge = c(9.84, 6.56), subcircle = 0.66))

Not quite sure what did it in the end! But I do think that the normalize_height function ought to carry out automatic indexing

closed 2020-10-26 by azh3

Jean-Romain, 2020-10-26:
Impossible to answer without a reproducible example. Did you tried to run the `pitfree` algorithm on a non-normalized dataset?