r-lidar/lidR #388: normalize_intensity - error with LAScatalog

lucas-johnson, 2020-11-06:
### Current Behavior
`normalize_intensity` works with a single las file, but fails with a las catalog. The error indicates that the algorithm provided (`range_correction(sensor_positions, Rs = 2000)`) is invalid, but it is the same for the single las file and the las catalog. 
```
> normalize_intensity(readLAS("~/Downloads/some_las/u_5550087800_2015.las"), range_correction(sensor_positions, Rs = 2000))
class        : LAS (v1.4 format 6)
memory       : 1.2 Gb 
extent       : 555000, 556500, 4878000, 4879500 (xmin, xmax, ymin, ymax)
coord. ref.  : +proj=utm +zone=18 +ellps=GRS80 +towgs84=0,0,0,0,0,0,0 +units=m +no_defs 
area         : 2.25 km²
points       : 12.96 million points
density      : 5.76 points/m²
> normalize_intensity(ctg, range_correction(sensor_positions, Rs = 2000))
Processing [=>------------------------------------------------------------------------------]   3% (1/38) eta:  2mError: Invalid function provided as algorithm.
> plot(ctg)
> range_correction(sensor_positions, Rs = 2000)
Object of class lidR algorithm
Algorithm for: intensity normalisation 
Designed to be used with: normalize_intensity 
Native C++ parallelization: no 
Parameters: 
 - elevation = Z <character>
 - f = 2.3 <numeric>
 - gpstime = gpstime <character>
 - Rs = 2000 <numeric>
 - sensor <SpatialPointsDataFrame>
```

### Desired Behavior
`normalize_intensity()` should work on  a las catalog just like it works on a single las file.  Or, since the algorithm used in both cases is the same, the error should describe a different issue (perhaps with my las catalog).

Jean-Romain, 2020-11-06:
Could please show a reproducible example with the exact error. There is no error in the code snippet you copied.

Jean-Romain, 2020-11-06:
Reproducible example

```r
LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las <- readLAS(LASfile)
sensor <- track_sensor(las, Roussel2020(pmin = 15))

ctg = readLAScatalog(LASfile)
opt_output_files(ctg) <- tempfile()
ctg <- normalize_intensity(ctg, range_correction(sensor, Rs = 2000))
#> Erreur : Invalid function provided as algorithm.

Jean-Romain, 2020-11-06:
After investigation, what is received is an `Extent` from `raster` and the method `normalize_intensity.LAScluster` is by-passed. This is weird. I'll investigate later. This will be fixed on Monday.

lucas-johnson, 2020-11-06:
Thank you for looking into it so quickly! Looking forward to hearing back from you.

Jean-Romain, 2020-11-06:
By the way, the methods does not require a buffer around the processed regions. Unless you want to process small chunks, you can write a simple `for` loop to process each file independently.

lucas-johnson, 2020-11-06:
Yes - thanks for the advice. I've written my own `.LAScatalog` and `.LAScluster` functions to work with `catalog_apply()`. This is working fine for now.

Jean-Romain, 2020-11-06:
Fixed. Thank you for reporting

commit c5873ae9 (2020-11-06): Fix #388
closed 2020-11-06 by Jean-Romain