r-lidar/lidR #389: Coordinate Reference System vanishes when it is wirtten out using `writeLAS` function

ShingObt, 2020-11-14:
Hi,

I am trying to save the clipped LAS file using `writeLAS` function. The las file I saved surely had CRS. But when I load the saved file onto R using `readLAS` function, CRS is shown as `NA`. 

```r
> pc=readLAS("thePointCloud.laz")
> pc
class        : LAS (v1.4 format 6)
memory       : 541.5 Kb 
extent       : 2267942, 2268061, 560393.9, 560513.6 (xmin, xmax, ymin, ymax)
coord. ref.  : +proj=lcc +lat_1=36.41666666666666 +lat_2=35.25 +lat_0=34.33333333333334 +lon_0=-86 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs  # coordinate reference system exists
area         : 11162.62 us-ft²
points       : 5.2 thousand points
density      : 0.47 points/us-ft²

> writeLAS(pc,"thePointCloud_saved.las",index=TRUE)
> readLAS("thePointCloud_saved.laz")
class        : LAS (v1.4 format 6)
memory       : 480 Kb 
extent       : 2267942, 2268061, 560393.9, 560513.6 (xmin, xmax, ymin, ymax)
coord. ref.  : NA # coordinate reference system is changed to NA
area         : 11162.62 units²
points       : 5.2 thousand points
density      : 0.47 points/units²
```
Is there any way to keep the CRS after `writeLAS`?
Or is it an intended change of the data? 

Thank you a lot in advance!

Jean-Romain, 2020-11-14:
Please share your file (or small chunk is enough) so I can reproduce. And please report the version of `rlas`, `lidR`, `sp`, `sf` as well as the version of `gdal` and `libproj`.

``` r
packageVersion("lidR")
#> [1] '3.0.5'
packageVersion("rlas")
#> [1] '1.3.7'
packageVersion("sp")
#> [1] '1.4.2'
packageVersion("sf")
#> [1] '0.9.5'
sf::sf_extSoftVersion()
#>           GEOS           GDAL         proj.4 GDAL_with_GEOS     USE_PROJ_H 
#>        "3.8.0"        "3.0.4"        "7.0.0"         "true"         "true"
```

<sup>Created on 2020-11-14 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

Jean-Romain, 2020-11-14:
I suspected an issue with LAS 1.4 so I tested myself and it worked. So far your bug is not confirmed

``` r
library(lidR)
las1 = readLAS("LAS 1.4 PDRF 6/e1054n0839.las", filter = "-keep_random_fraction 0.01")
las1
#> class        : LAS (v1.4 format 6)
#> memory       : 15.7 Mb 
#> extent       : 1054000, 1055000, 839000, 840000 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NAD83 (2011) / Conus Albers 
#> area         : 1 km²
#> points       : 178.9 thousand points
#> density      : 0.18 points/m²
f = tempfile(fileext = ".las")
writeLAS(las1, f)
las2 = readLAS(f)
las2
#> class        : LAS (v1.4 format 6)
#> memory       : 15.7 Mb 
#> extent       : 1054000, 1055000, 839000, 840000 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NAD83 (2011) / Conus Albers 
#> area         : 1 km²
#> points       : 178.9 thousand points
#> density      : 0.18 points/m²
```

<sup>Created on 2020-11-14 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

ShingObt, 2020-11-15:
Thank you. 

My LAZ file is 
[pointcloud.laz.zip](https://github.com/Jean-Romain/lidR/files/5543110/pointcloud.laz.zip)

The LAZ file does not have any CRS when it was downloaded that I assigned CRS by using metadata.

``` R
> pc=readLAS("pointcloud.laz")# no crs assigned in the original data # no CRS is assigned.
> pc
class        : LAS (v1.4 format 6)
memory       : 28.8 Mb 
extent       : 2215400, 2216600, 526400, 527600 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 1.13 kunits²
points       : 328 thousand points
density      : 0.29 points/units²
> pc@proj4string=crs("+proj=lcc +lat_1=36.41666666666666 +lat_2=35.25 +lat_0=34.33333333333334 +lon_0=-86 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs")# assign proj4 strings by matching metadata.
> lazclip=clip_circle(pc,2216000,527000,radius=60)# for simplification I directly set the xcenter and ycenter as number, but it is converted from latlon using spTransform.
> lazclip
class        : LAS (v1.4 format 6)
memory       : 271.1 Kb 
extent       : 2215940, 2216060, 526940.8, 527059.4 (xmin, xmax, ymin, ymax)
coord. ref.  : +proj=lcc +lat_1=36.41666666666666 +lat_2=35.25 +lat_0=34.33333333333334 +lon_0=-86 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs 
area         : 11069.44 us-ft²
points       : 2.9 thousand points
density      : 0.26 points/us-ft²
> writeLAS(lasclip,"pointcloud_clip.laz")
> readLAS("pointcloud_clip.laz") # No CRS
class        : LAS (v1.4 format 6)
memory       : 28.8 Mb 
extent       : 2215400, 2216600, 526400, 527600 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 1.13 kunits²
points       : 328 thousand points
density      : 0.29 points/units²

``` 

Versions for the packages are as follow. 

```r
> packageVersion("lidR")
[1] ‘3.0.4’
> packageVersion("rlas")
[1] ‘1.3.7’
> packageVersion("sp")
[1] ‘1.4.4’
> packageVersion("sf")
[1] ‘0.9.6’
> packageVersion("libproj")
[1] ‘7.1.0.3’
> sf::sf_extSoftVersion()
          GEOS           GDAL         proj.4 GDAL_with_GEOS     USE_PROJ_H 
       "3.6.2"        "2.2.3"        "4.9.3"         "true"        "false" 
```
GDAL version was 2.2.3.
```linux
(base) obata:~$ gdalinfo --version
GDAL 2.2.3, released 2017/11/20
```

So comparing to your versions, external dependencies versions of the libraries linked to sf were older than the versions of the packages available in my environment.

Jean-Romain, 2020-11-15:
Ha, you hid some aspects of your problem...

See `?lidR::projection` and especially the section "Details".

closed 2020-11-15 by Jean-Romain

Jean-Romain, 2020-11-16:
```r
library(lidR)
las = readLAS("~/Documents/ALS data/Fichiers las divers/LAS 1.4 PDRF 6/pointcloud.laz")
#> Warning: There are 2 points flagged 'withheld'.
projection(las) = crs("+proj=lcc +lat_1=36.41666666666666 +lat_2=35.25 +lat_0=34.33333333333334 +lon_0=-86 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs")# assign proj4 strings by matching metadata.
las
#> class        : LAS (v1.4 format 6)
#> memory       : 28.8 Mb 
#> extent       : 2215400, 2216600, 526400, 527600 (xmin, xmax, ymin, ymax)
#> coord. ref.  : +proj=lcc +lat_0=34.3333333333333 +lon_0=-86 +lat_1=36.4166666666667 +lat_2=35.25 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs 
#> area         : 1.13 kus-ft²
#> points       : 328 thousand points
#> density      : 0.29 points/us-ft²
wkt(las)
#> [1] "PROJCRS[\"unknown\",\n    BASEGEOGCRS[\"unknown\",\n        DATUM[\"Unknown based on GRS80 ellipsoid\",\n            ELLIPSOID[\"GRS 1980\",6378137,298.257222101,\n                LENGTHUNIT[\"metre\",1],\n                ID[\"EPSG\",7019]]],\n        PRIMEM[\"Greenwich\",0,\n            ANGLEUNIT[\"degree\",0.0174532925199433],\n            ID[\"EPSG\",8901]]],\n    CONVERSION[\"unknown\",\n        METHOD[\"Lambert Conic Conformal (2SP)\",\n            ID[\"EPSG\",9802]],\n        PARAMETER[\"Latitude of false origin\",34.3333333333333,\n            ANGLEUNIT[\"degree\",0.0174532925199433],\n            ID[\"EPSG\",8821]],\n        PARAMETER[\"Longitude of false origin\",-86,\n            ANGLEUNIT[\"degree\",0.0174532925199433],\n            ID[\"EPSG\",8822]],\n        PARAMETER[\"Latitude of 1st standard parallel\",36.4166666666667,\n            ANGLEUNIT[\"degree\",0.0174532925199433],\n            ID[\"EPSG\",8823]],\n        PARAMETER[\"Latitude of 2nd standard parallel\",35.25,\n            ANGLEUNIT[\"degree\",0.0174532925199433],\n            ID[\"EPSG\",8824]],\n        PARAMETER[\"Easting at false origin\",1968500,\n            LENGTHUNIT[\"US survey foot\",0.304800609601219],\n            ID[\"EPSG\",8826]],\n        PARAMETER[\"Northing at false origin\",0,\n            LENGTHUNIT[\"US survey foot\",0.304800609601219],\n            ID[\"EPSG\",8827]]],\n    CS[Cartesian,2],\n        AXIS[\"(E)\",east,\n            ORDER[1],\n            LENGTHUNIT[\"US survey foot\",0.304800609601219,\n                ID[\"EPSG\",9003]]],\n        AXIS[\"(N)\",north,\n            ORDER[2],\n            LENGTHUNIT[\"US survey foot\",0.304800609601219,\n                ID[\"EPSG\",9003]]]]"
epsg(las)
#> Warning: This LAS object stores the CRS as WKT. 0 returned; use 'wkt()' instead.
#> [1] 0
f = tempfile(fileext = ".las")
writeLAS(las, f)
las2 = readLAS(f)
#> Warning: There are 2 points flagged 'withheld'.
las2
#> class        : LAS (v1.4 format 6)
#> memory       : 28.8 Mb 
#> extent       : 2215400, 2216600, 526400, 527600 (xmin, xmax, ymin, ymax)
#> coord. ref.  : +proj=lcc +lat_0=34.3333333333333 +lon_0=-86 +lat_1=36.4166666666667 +lat_2=35.25 +x_0=600000 +y_0=0 +ellps=GRS80 +units=us-ft +no_defs 
#> area         : 1.13 kus-ft²
#> points       : 328 thousand points
#> density      : 0.29 points/us-ft²
```

ShingObt, 2020-11-17:
Thank you! It worked well in my environment.

rphillips91, 2021-08-04:
Would this be a similar issue - should I make a new issue specific is this? (rookie git issuer here) 
https://stackoverflow.com/questions/68644010/las-dual-projection-not-recognized-by-r-postgis-3-0-proj-db-error-raised

Jean-Romain, 2021-08-04:
This works with a "SingleCRS" message

```r
wkt = 'COMPD_CS["NZGD2000 / New Zealand Transverse Mercator 2000 + NZVD2016 height",PROJCS["NZGD2000 / New Zealand Transverse Mercator 2000",GEOGCS["NZGD2000",DATUM["New_Zealand_Geodetic_Datum_2000",SPHEROID["GRS 1980",6378137,298.257222101,AUTHORITY["EPSG","7019"]],AUTHORITY["EPSG","6167"]],PRIMEM["Greenwich",0,AUTHORITY["EPSG","8901"]],UNIT["degree",0.0174532925199433,AUTHORITY["EPSG","9122"]],AUTHORITY["EPSG","4167"]],PROJECTION["Transverse_Mercator"],PARAMETER["latitude_of_origin",0],PARAMETER["central_meridian",173],PARAMETER["scale_factor",0.9996],PARAMETER["false_easting",1600000],PARAMETER["false_northing",10000000],UNIT["metre",1,AUTHORITY["EPSG","9001"]],AXIS["Northing",NORTH],AXIS["Easting",EAST],AUTHORITY["EPSG","2193"]],VERT_CS["NZVD2016 height",VERT_DATUM["New Zealand Vertical Datum 2016",2005,AUTHORITY["EPSG","1169"]],UNIT["metre",1,AUTHORITY["EPSG","9001"]],AXIS["Gravity-related height",UP],AUTHORITY["EPSG","7839"]]]'
sp::CRS(SRS_string = wkt)
#> proj_crs_get_datum: Object is not a SingleCRS
#> proj_crs_get_datum: Object is not a SingleCRS
#> CRS arguments:
#>  +proj=tmerc +lat_0=0 +lon_0=173 +k=0.9996 +x_0=1600000 +y_0=10000000 +ellps=GRS80 +units=m +vunits=m +no_defs 
```
If `readLAS()` fails at reading the file please open an issue here with a link to the file
If `readLAS()` read the files properly and you can reproduce your message with `sp` or `sf` only please ask on gis.stackexhange. I'm not an expert of CRS. Maybe ask on `rgdal` repo.
For all other case we will see. Anyway don't ask `lidR` question on stackoverflow. There is no `lidr` tag and no user there. You are unlikely to get any anwser.