r-lidar/lidR #390: Variable Search Parameter in li2012 function

azh3, 2020-11-18:
I was wondering if there was a way to implement a variable search window within the li2012 function? For example, I might set the variable search radius according to the following function, 'ws'. 

```
ws <- function(x){ifelse(x>0, return((0+8.8390*log(x))/2), return(1))}
alg <- lidR::li2012(dt1 = 5, dt2 = ws, R = ws, Zu = 15, hmin = 15, speed_up = 32)
```

When I try plugging this into the equation, I get the following error: 

`Error: ws is not a number.`

And just to be sure I am understanding this function and the literature correctly, dt2 refers to tree spacing of points greater than 15m while dt1 refers to the minimum tree spacing of points less than 15m? While R refers to the actual search radius that is applied to every point?

Jean-Romain, 2020-11-18:
All parameters are number,s this is how the algorithm is described in the paper. I won't change that otherwise the algorithm will no longer be the one described in the paper.

I don't think your understanding of the algorithm is fully correct. Your description being short I'd say yes it is correct except for `R`. In the lidR's implementation (which is based on my understanding of the paper) R is the radius used to tag points as local maxima. If a point is a local maxima then rule 1 applies otherwise rule 2. And I don't understand what do you mean when you are talking about *fitted radius of a tree crown*. There is no such step in the methods. The methods processes each points and classifies in two groups then it redo until no point remain. There is no "crown" or "fit" of any kind.

azh3, 2020-11-18:
According to Li et al (2012) "Within a search radius R, the highest point is called local maximum, and any other lower point is called non-maximum." While they note that the segmentation result is not sensitive to search radius, my goal here is to define the local maxima based on an established tree-height diameter relationship, similar to what one might do using the lmf() function for finding treetops. So I am going to assume that, at this point, there is no way to define a variable window size within the li2012() algorithm because it would deviate from the algorithm defined in the paper. I was hoping to do something similar within the max_cr_factor argument to the silva2016 function, and have received the same error message. If I wanted to call the ws algorithm (above) within either of these functions, is there a clear path to modifying the function to do so?

Jean-Romain, 2020-11-18:
In li's algorithm the local maxima do not necessarily correspond to treetops. They can be e.g. branches. More importantly the local maxima are not the seeds of the trees. Consequently transferring what you found with `lmf()` to `li2012()` does not make sense here. 

> If I wanted to call the ws algorithm (above) within either of these functions, is there a clear path to modifying either of these functions to do so?

Yes here

https://github.com/Jean-Romain/lidR/blob/2833fcf1215f6726498ad65c97f0a2ced1b39313/src/LAS.cpp#L964

Replace `NumericVector::create(R)` by a vector that contains the radius you want to use for each point i.e. the output of `ws()` for each `Z`

azh3, 2020-11-18:
Jean-Romain, thank you for your help, I will give it a shot.

closed 2020-11-18 by azh3