r-lidar/lidR #391: Potential Bug In track_sensor()

lucas-johnson, 2020-11-19:
Similar to: https://github.com/Jean-Romain/lidR/issues/336. Unrealistic ranges are being computed due to `gpstime` matching. I'm not sure if my data is bad, or if the `track_sensor()` results are bad. My assumption is that the sensor `gpstime` should match (relatively speaking) the `gpstime` of the the las which it overlaps. Perhaps I am using too large of buffers (100m on 2.25 km² area tiles)? I doubt this is the problem. The data is very large (catalog ~30GB total) so it will be challenging to share a complete reprex. 

Here are the tiles with the sensor positions overlaid (in black), and the mismatching locations. The green point is the lidar point, and the red point is the matching sensor position.
[image]

Here is sensor positions colored by `gpstime`. This seems odd to me, though I'm not sure it should create unrealistic range.
[image]

Example code:
```{r}
ctg <- readLAScatalog("~/Downloads/chunk_2_raw_11182020")
opt_chunk_buffer(ctg) <- 100
# Filter low/high noise
opt_filter(ctg) <- "-drop_withheld -drop_class 7 18"
sensor_positions <- track_sensor(ctg, Roussel2020(), multi_pulse = T)
range_corrected <- normalize_intensity(ctg, range_correction(sensor_positions, 2260))
#> Processing [========================================================================>---------]  89% (39/44) eta: 48sAn high range R has been computed relatively to the expected average range Rm = 1841
#> Point number 17186276 at (x,y,z,t) = (578460.10, 4804968.93, 695.17, 114705070.00)
#> Matched with sensor between (577549.19, 4802867.12, 2575.20, 114705296.50) and (577546.65, 4802902.00, 2578.17, 114705297.00)
#> The range computed was R = 17912.55
#> Check the correctness of the sensor positions and the correctness of the gpstime either in the point cloud or in the sensor positions.
#> Processing [==========================================================================>-------]  91% (40/44) eta: 39sAn error occurred when processing the chunk 40. Try to load this chunk with:
#> chunk <- readRDS("/var/folders/_c/bcw1j0251k12c8hqxpqpb7nr0000gn/T//Rtmp4NqbVJ/chunk40.rds")
#>  las <- readLAS(chunk)
#> Unrealistic range: see message above
#> Warning messages:
#> 1: Invalid data: 3 points with a number of returns equal to 0 found. 
#> 2: Invalid data: 5 points with a 'return number' greater than the 'number of returns'. 
#> 3: Invalid data: 169 points with a number of returns equal to 0 found. 
#> 4: Invalid data: 225 points with a 'return number' greater than the 'number of returns'. 

```

Jean-Romain, 2020-11-19:
I think you should be able to reproduce by clipping e.g a 10 m disc around (578460.10, 4804968.93). I was very inspired the day I added this security test *just in case*. But I can hardly guess what is going wrong with that. I can see that the gpstime matching is incorrect.

```r
las = clip_disc(ctg, 578460.10, 4804968.93, 10)
normalize_intensity(las, range_correction(sensor_positions, 2260))
```

lucas-johnson, 2020-11-19:
[lidr_reprex_11192020.zip](https://github.com/Jean-Romain/lidR/files/5566932/lidr_reprex_11192020.zip)

Jean-Romain, 2020-11-19:
Ok, here what I found. You have 3 points (669, 670, 671) from a single pulse with a PointSourceID = 100 

```r
las@data[PointSourceID == 100, .(X,Y,Z, gpstime, ReturnNumber, NumberOfReturns)]
#>          X       Y      Z   gpstime ReturnNumber NumberOfReturns
#> 1: 578460.1 4804969 695.17 114705070            1               3
#> 2: 578460.8 4804968 681.59 114705070            2               3
#> 3: 578461.4 4804968 671.34 114705070            3               3
```
But you don't have any sensor position with a PointSourceID = 100. Those 3 points seems to be the 3 unique points from the swath number 100. You can try to find some other ones in your tile but I pretty sure they are isolated and there is not other ones. Or maybe few other ones spread far from each other. 

The 3 points of swath 100 being from a single pulse there is no way to compute a sensor position for this swath and thus the swath 100 is missing in the shapefile. When trying to normalize points 669 there is no gpstime match. The algo tries to find the next best match but it is irrelevant and thus an error is thrown to tell you to check the correctness of the positions. And here we are.

If swath 100 only contains those 3 points and you are ok for removing swath 100 just use:

```r
opt_filter(ctg) <- "-drop_point_source 100"
```

But first you should investigate a little bit more on swath 100 to figure out what it is, how many points it contains and so on...

lucas-johnson, 2020-11-19:
Thank you! I compiled all of the data from the catalog into one giant `data.table`. Here is what I found with regard to `PointSourceID == 100`:

```r
nrow(catalog_census)
#> [1] 1060753974
nrow(catalog_census[PointSourceID == 100])
#> [1] 122
nrow(catalog_census[PointSourceID == 100 & !Classification %in% c(7, 18)])
#> [1] 49
nrow(catalog_census[PointSourceID == 100 & !Classification %in% c(7, 18) & Overlap_flag == FALSE])
#> [1] 14
```
Here is the spatial arrangement of the 49 points where `PointSourceID == 100 &  !Classification %in% c(7, 18)`:
[image]

Given the very low number of points with `PointSourceID == 100` in this specific catalog, I think it is safe to leave them out. But why are they there (I'm sure nobody really knows)? Their odd spatial arrangement and the fact that there are so few of them lead me to believe they are completely erroneous.

I have questions about the implications on a larger scale. Do `PointSourceID`s refer to unique flight paths/swaths in a given LiDAR coverage? I.e. is a `PointSourceID` reused in another tile from the same coverage? I'm wondering if it would be safe to drop `PointSourceID == 100` from all tiles in the LiDAR coverage (currently looking at 66 of some 2000 of them)...

Jean-Romain, 2020-11-19:
> Do PointSourceIDs refer to unique flight paths/swaths in a given LiDAR coverage?

PointSourceID is expected to record a single swath. `track_sensor()` assumes it does. But indeed it can be incorrectly populated and honestly I don't know the consequences for `track_sensor()` because there is virtually an infinite number of way to wrongly populate this attribute. You just found a limit case I would never have thought of. Another case I have in mind is if your collection comes from two different coverage/contract and there are overlap in gpstime or point sources.

If two swath share the same ID but are far from each other it is not an issue because for a given processed chunk they won't be seen both at the same time and no confusion can happen. Then in `normalize_intensity` the point source is no longer used. Only gpstime matters. Consequently overlaps in gpstime are more critical. The swath number is not actually strictly required we could work without it. But I enforced to work by swath to add another layer of safety. And your error proved I was right.

lucas-johnson, 2020-11-19:
Thanks for the clarification - this makes sense to me. So if I don't remove `PointSourceID == 100` from the catalog, but then set `opt_stop_early(ctg) <- FALSE`, I believe `normalize_intensity()` will entirely skip the tiles where the error occurs (testing now). Instead, I'm wondering what you think about adding an option to drop points where the `Unrealistic range` is computed? Like `normalize_intensity(ctg, range_correction(sensor_positions, 2260, drop_bad_range_pts = TRUE`)). This assumes the points where unrealistic ranges are computed are few and erroneous.

Jean-Romain, 2020-11-19:
> `opt_stop_early(ctg) <- FALSE`, I believe `normalize_intensity()` will entirely skip the tiles where the error occurs (testing now)

yes

> Like normalize_intensity(ctg, range_correction(sensor_positions, 2260, drop_bad_range_pts = TRUE)). This assumes the points where unrealistic ranges are computed are few and erroneous.

I don't think it is a good idea because bad range can occur for good reasons. For example I have a file given by the data provider that records the position of the sensor with incorrect gpstime (a factor of 1000 between the point cloud and the sensor records). This would give 0 point in output but no error. In you case use `"-drop_point_source 100"`. This is likely the best option.

I'm open to suggestion but you have already experienced many issues with these two functions. And I experienced many others. As I said there is virtually an infinite number of way to wrongly populate the dataset and I cannot think of every possibilities. So I prefer the function to fail often so we can investigate. By systematically investigating I can improve the function. In you case I'll try to add a warning in `track_sensor()`. Something like

```
Warning: some swaths (PointSourceID == xxx,yyy,zzz) contains points but are not associated to any sensor position. Something unexpected happened with these swath
```

lucas-johnson, 2020-11-19:
Updated, my previous post to fix some problems with the previous results... 

I was thinking the option to drop "bad range" points would produce a warning without failing, but fair enough – I see your point. There are just too many ways that problems can occur.  I think your planned warning with `track_sensor()` would be very helpful. Thanks for taking a look at all of this.

lucas-johnson, 2020-11-19:
Another thought inline with the `track_sensor()` warning would be a function to return the values of `PointSourceIDs` that do not match any sensor positions. Might be nice to access directly rather than just viewing/parsing the warning messages.

Jean-Romain, 2020-11-19:
```r
library(lidR)
las = readLAS("reprex_las_11192020.las")
trk = track_sensor(las, Roussel2020(pmin = 2))
#> Warning: Some swaths present in the point cloud (PointSourceID = [100]) did
#> not produce any sensor location
comment(trk)
#> [1] "100"
```
--------

Not related but please see my message in the [chat of gis.stackechange](https://chat.stackexchange.com/rooms/116345/discussion-between-jrr-and-lucas) (it seems we do not get notification for chat message). It would be really appreciated if you could report a minimal reproducible example here.

Jean-Romain, 2020-11-19:
By the way

```r
point_source_100 <- lapply(ctg$filename, function(x) {
    a_las <- readLAS(x, select = "xypc", filter = "-drop_withheld -keep_point_source 100")
    return(a_las@data)
})
ps100_dt <- rbindlist(point_source_100)
```

can be written

```r
readLAS(ctg, select = "xypc", filter = "-drop_withheld -keep_point_source 100")
```

:wink:

closed 2020-11-19 by Jean-Romain