r-lidar/lidR #393: Find corrupted point for track_sensor

Jean-Romain, 2020-11-23:
Following numerous issues opened by @lucas-johnson (#392, #391, #388, #336, #327), add a function: `diagnose_invalid_pulse()`.

This could  return a table with the index of the point + the points + a comment explaining why a pulse is invalid.

closed 2021-05-27 by Jean-Romain

candelas762, 2022-07-19:
Hi. Was this ever included in any version of ```lidR```? After having read what I think are all questions and issues with ```track_sensor()``` so far, specially #392 , #396 and #595, I still can't find the reason why some parts of catalogs retrive the error ```Error: After keeping only first and last returns of multiple returns pulses, X pulses still have more than 2 points. This dataset is corrupted and gpstime is likely to be invalid.```. I wanted to try dropping those points if they were not many.

Jean-Romain, 2022-07-19:
Feature never implemented. Can you share a file with an error ?

candelas762, 2022-07-20:
Sure. You can download the file from [here](https://drive.google.com/file/d/1F7TMq2SMhQix_FzIHG60SiyAio8GVN50/view?usp=sharing). The "retile" file is is a part of  the bigger ALS file read as a ```catalog``` with ```opt_chunk_buffer(ctg) <- 0.5``` and ```opt_chunk_size(ctg) <- 250```. I isolated that part because is small (easy to share) and still have the error. I also share the whole file in case these intensisty functions need to work with bigger files. 

```
las <- readLAS("C:/retile_578250_6677250.las", filter = "-drop_class 7")

las <- filter_duplicates(las) # I think that track_sensor() does this automatically 
las <-  filter_firstlast(las)
sensor <- track_sensor(las, Roussel2020(pmin = 500))
#> Error: After keeping only first and last returns of multiple returns pulses, 44 pulses still have 
#> more than 2 points. This dataset is corrupted and gpstime is likely to be invalid.

```

Jean-Romain, 2022-07-20:
I see

```
 - Checking gpstime incoherances
    ✗ 80464 pulses (points with the same gpstime) have points with identical ReturnNumber
```

Label the incorrect ones

```r
las = filter_firstlast(las)
t = las@data[, .N, by = gpstime]
err = t[N > 2]
las@data$err = las$gpstime %in% err$gpstime
```

Plot the incorrect ones

```r
plot(las, color = "err")
```

Show one of the inccorect ones

```r
tt = err$gpstime
las@data[gpstime == tt[1]]
#>          X       Y      Z   gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag
#> 1: 578410.5 6677400 255.02 114610821         4            1               2                 0                0              1          FALSE         FALSE
#> 2: 578412.2 6677401 243.03 114610821        10            2               2                 0                0              1          FALSE         FALSE
#> 3: 578411.0 6677400 251.70 114610821       127            1               1                 0                0              1          FALSE         FALSE
#>    Withheld_flag ScanAngleRank UserData PointSourceID  err
#> 1:         FALSE            -8        1             1 TRUE
#> 2:         FALSE            -8        1             1 TRUE
#> 3:         FALSE            -8        1             1 TRUE
```

Plot those 3 points

```r
sub = las[las$gpstime == tt[1]]
plot(sub, size = 4)
```

The 3 points are perfectly aligned. I'd say they are coming from the same pulse but numbering is incorrect. This pulse is a 3 returns pulse and points should have 1 2 3 in return number and 3 3 3 in number of return. You have probably 80000+ point of this kind. You dataset is incorrectly populated and the function is working as expected.

Jean-Romain, 2022-07-20:
The fastest way is

```r
library(lidR)
library(data.table)
las <- readLAS("~/Téléchargements/retile_578250_6677250.las", filter = "-drop_class 7")
setorder(las@data, gpstime, UserData, Z)
las@data[, `:=`(ReturnNumber = 1:.N, NumberOfReturns = .N), by = .(gpstime, UserData)]
las = las_update(las)
```

Notice that you missed to processed by user data because you have a multi pulse emission device. This is also why `las_check()` still report gpstime error. It does not accounts for multi pulse that have the same gpstime but different UserData.
```r
las = filter_firstlast(las)
t = las@data[, .N, by = .(gpstime, UserData)]
setorder(t, N)
err = t[N > 2]
err
#> Empty data.table (0 rows and 3 cols): gpstime,UserData,N
```

candelas762, 2022-07-21:
Amazing that line of code. Thanks a lot.

I included those lines in case the error came up and compared with the actual flightlines shapefile. It aparantley solves the corrupted points and the sensor is well-positioned. 

In the image you can see the ```las``` file read as a ```catalog```, the flightlines colored by ```ID```,  and the black points are the sensor positions calculated for the ```catalog``` and the red circles are the sensor positions calculated just for the first retile read as a ```las```, which is pink colored.
[image]

It seems like the bigger the data, the better positioned the sensor gets.

I was wondering if given the case where you have access to the actual flight lines, it was possible to make your own "```track_sensor()```" result.

Jean-Romain, 2022-07-21:
> It seems like the bigger the data, the better positioned the sensor gets.

Of course. With a tiny are you have very few points to resolve the interpolation problem. But the improvement is not very big actually

> I was wondering if given the case where you have access to the actual flight lines, it was possible to make your own "track_sensor()" result.

If the shapefile is made of lines I guess it is not associated with timestamps. We need a time information to match the point with the position of the sensor. You can sample the lines into discrete points but if the points are not associated with corresponding gpstime it won't work.