r-lidar/lidR #395: range_correction error: sensor positions do not overlap with point cloud

lucas-johnson, 2020-11-25:
It seems that `normalize_intensity()` with `range_correction()` enforces that sensor positions intersect each chunk/tile. Should this be enforced? Here is my example:

```r
ctg
#> class       : LAScatalog (v1.4 format 6)
#> extent      : 358631.3, 366000, 4719000, 4732500 (xmin, xmax, ymin, ymax)
#> coord. ref. : NAD83(2011) / UTM zone 18N + NAVD88 height - Geoid12B (metres) (with axis order normalized for visualization) (with axis order normalized for visualization) 
#> area        : 55.18 km²
#> points      : 249.25 million points
#> density     : 4.5 points/m²
#> num. files  : 26 

opt_chunk_buffer(ctg) <- 100
opt_filter(ctg) <- "-drop_user_data 5"
sensor_positions <- track_sensor(ctg, Roussel2020(), multi_pulse = T, thin_pulse_with_time = 0)
normalize_intensity(ctg, range_correction(sensor_positions, 2000))
#> Error: Point-cloud and sensor positions do not overlap.
plot(sensor_positions, add = T)
```
Progress plot with `sensor_positions` overlaid:
[image]

Jean-Romain, 2020-11-25:
Honestly I added this constraint to protect against unexpected behavior without being sure of the side effects. From the beginning you are the guinea pig for these functions. They are so many possible unexpected problem. 

In your case this constraint does not seem legitimate and should be relaxed. 

I think I added this constraint to protect against the case where the sensor data do no cover the whole point cloud. This may happen if the sensor data are coming from a third party source. Anyway I'll relax that.

lucas-johnson, 2020-11-25:
Yes  I understand I'm the guinea pig for a lot of these things, and I understand that you could not have predicted all of the problems/edge cases/intricacies. I appreciate your work here, and your agility on fixing issues and making updates. I do think that some spatial constraint on the sensor positions and the point cloud makes sense. Perhaps the `Unrealistic range computed` error implicitly handles these kinds of issues?

Jean-Romain, 2020-11-25:
Done

closed 2020-11-25 by Jean-Romain

Jean-Romain, 2020-11-25:
> I do think that some spatial constraint on the sensor positions and the point cloud makes sense.

I do agree

> Perhaps the Unrealistic range computed error implicitly handles these kinds of issues?

Yes I should put the unrealistic range test in `range_correction()`. Currently there is no such test. Sensor data are expected to be valid.

lucas-johnson, 2020-11-25:
Ohh I thought you had that test in there: https://github.com/Jean-Romain/lidR/issues/391#issuecomment-730366511

Jean-Romain, 2020-11-25:
Ho you're right there a test ! :+1: