r-lidar/lidR #396: Feature request: filter duplicates in track_sensor

lucas-johnson, 2020-11-25:
Potential fix for this kind of problem: https://github.com/Jean-Romain/lidR/issues/392#issuecomment-732350794

It might be nice to filter out duplicates in the `track_sensor()` function. Filtering these duplicates will not change the underlying data but will only be used to compute the sensor positions so it seems pretty safe. 

```r
track_sensor.LAScluster <- function(las, algorithm, extra_check = TRUE, thin_pulse_with_time = 0.001, multi_pulse = FALSE)
{
  x <- readLAS(las)
  if (is.empty(x)) return(NULL)
  x <- filter_duplicates(x)
  pos  <- track_sensor(x, algorithm, extra_check, thin_pulse_with_time, multi_pulse)
  return(pos)
}
```

Jean-Romain, 2020-11-25:
Do agree

Jean-Romain, 2021-05-27:
Done

commit 5f4f8d3f (2021-05-27): Filter duplicate (close #396)
closed 2021-05-27 by Jean-Romain