r-lidar/lidR #397: Feature Request: Populate pulse/beam id

lucas-johnson, 2020-11-25:
### Problem
The initial problem is well documented [here](https://gis.stackexchange.com/questions/354769/split-las-data-from-multi-beam-sensors).  In short, I would like to be able to use `track_sensor()` on LiDAR data that is multi-beam or multi-pulse (two distinct pulses are emitted with the same `gpstime`). You have nicely added the `multi_pulse` option for `track_sensor()` however this only works if the beam/pulse is properly identified in the `UserData` field of the las file/catalog. So, we have no way to use `track_sensor()` on data with an unpopulated or improperly populated `UserData` field. 

### Solution
Add a function to populate the `UserData` field by grouping points on `gpstime` and `NumberOfReturns`. The function needs to be applied with buffers, since `track_sensor()` needs to be applied with buffers and that is where things get a bit tricky. We cannot properly identify "beam/pulse ids" for paired pulses that have the same `NumberOfReturns` so we must flag these and filter them out before `track_sensor()`. Here is some example code - it works for me so far, but is a bit slow. I'm sure there are some more elegant/efficient solutions to be found, but this is the one that first came to mind:

```r
add_beam_id.LAScluster <- function(las) {
  las <- readLAS(las)
  if (is.empty(las)) return(NULL)
  
  # Order for speed.
  las@data <- las@data[order(gpstime, NumberOfReturns)]
  
  # Split groups by NumberOfReturns
  las@data[,
           flag := (1 * (rleid(NumberOfReturns) - 1 > 0)),
           by = "gpstime"]
  
  # Find split beams that span several buffer zones
  las@data[split == T, split := any(flag * buffer) > 0, by = "gpstime"] 
  
  # Flip flag so buffer zones are flagged as 0 and non-buffer zones are flagged 
  # with numbers reflecting the buffer. Do this because we won't write the flags
  # in the buffer zones.
  las@data[split == T, flag := abs(buffer - max(buffer)), by = "gpstime"] 
  las@data[split == T & flag > 0, flag:= max(flag), by = "gpstime"]
  
  # Find groups that can't be split by NumberOfReturns
  las@data[, problem := .N > NumberOfReturns, by = c("gpstime", "flag")]
  # Find groups that span more than 2 buffer zones. These are too tricky
  las@data[problem == F, problem := length(unique(flag)) > 2, by = "gpstime"]
  # Fill in gaps in a groupo
  las@data[, problem := any(problem), by = "gpstime"]
  
  # Mark groups that we couldn't split
  las@data[problem == T, flag := 5]
  
  # Cleanup
  las@data$UserData <- as.integer(las@data$flag)
  las@data[, problem := NULL]
  las@data[, flag := NULL]
  las <- filter_poi(las, buffer == 0)
  
  return(las)
}

add_beam_id.LAScatalog <- function(ctg) {
  options <- list(automerge = TRUE)
  ctg_w_beam <- catalog_apply(ctg, add_beam_id.LAScluster, .options = options)
  return(ctg_w_beam)
}
```

### Tests
I haven't tested it extensively, but on a ~5500 hectare catalog, with 26 tiles and 249.25 million points. With this case only ~.44 million points are classified as `5` indicating that we could not accurately split them for one reason or another. We can then successfully run: 

```r
opt_filter(ctg) <- "-drop_user_data 5
opt_chunk_buffer(ctg) <- 100
opt_output_files(ctg) <- ""
sensor_positions <- track_sensor(ctg, Roussel2020(), multi_pulse = T, thin_pulse_with_time = 0)
```
*** Note - If don't set `thin_pulse_with_time = 0` the function works but we get warnings:
```r
#> 1: Some swaths present in the point cloud (PointSourceID = [49]) did not produce any sensor location. 
```

### Summary
The solution is by no means perfect, but it aims to solve a problem that has no perfect solutions.  Certainly it needs more testing. But I wanted to see what your thoughts are.

Jean-Romain, 2020-11-25:
Your idea is good and you identified the limits (impossible to identify beams with the same number of returns). Sadly I think it is a very specific problem that does not have a satisfying generic solution. We cannot just write a function `identify_beam` that attributes a value to each beam. This function should also remove the points that cannot be identified. This might be good for you but I don't think it is reliable enough to be exposed to all users.

Consequently I think you should keep that for your purpose. If you really want to share that you can write a wiki page. You have an access to the LAScatalog processing engine so you can build your own modified `sensor_tracking()` function. This is really for this kind of usage that the engine has been built and documented.

That being said your function does not need a buffer. You can apply it to any point cloud.  This simplified version should do the job

```r
identify_and_filter_beam <- function(las) 
{

  # Order for speed.
  las@data <- data.table::setorder(las@data, gpstime, NumberOfReturns)
  
  f = function(NoR)
  {
    userdata = (1 * (data.table::rleid(NoR) - 1 > 0))
    splitted = any(userdata >= 1)
    return(list(UserData = userdata, flag = splitted))     
  }

  # Split groups by NumberOfReturns
  beams = las@data[, f(NumberOfReturns), by = gpstime]
  
  las@data[["UserData"]] = beams[["UserData"]]
  las = filter_poi(las, beams$flag == TRUE)
  
  return(las)
}
```

lucas-johnson, 2020-11-25:
Understood that the reliability is not there to create a generic function within the package. I will certainly hold on to my own personal version using the nice tools in the catalog engine.

Though I disagree on the buffer problem. `gpstime` groups can be split between files (I have seen this). If you don't use buffers when creating the beam id, you may miss some of the group and both parts will separately be marked `UserData = 0` in each file. When track_sensor() reads in the data with buffers you suddenly have a `gpstime` group with two `ReturnNumber = 1` because `UserData = 0` for both.

Jean-Romain, 2020-11-25:
I mean, the function does not need to be written in such a way that it considers and processes the variable named  `buffer` has you did. Just write a simple function without considering the buffer and load the data with a buffer when processing the chunks.

lucas-johnson, 2020-11-25:
Ok - I think I understand what you are saying. However I still stand by my point that we need to consider the buffers...

Here is the example I get when using a modified version of  [your function](https://github.com/Jean-Romain/lidR/issues/397#issuecomment-733903693): 
```r
test_jrr_func <- function(las) {
  las <- readLAS(las)
  if (is.empty(las)) return(NULL)
  las@data <- data.table::setorder(las@data, gpstime, NumberOfReturns) 
  
  f = function(NoR)
  {
    userdata = (1 * (data.table::rleid(NoR) - 1 > 0))
    flag <- all(userdata == 0) & any(length(userdata) > NoR)
    return(list(UserData = userdata, flag = rep(flag, length(userdata))))
  }
  
  beams = las@data[, f(NumberOfReturns), by = gpstime]
  las@data[["UserData"]] = as.integer(beams[["UserData"]])
  las = filter_poi(las, beams$flag == FALSE)
  las = filter_poi(las, buffer == 0)
  return(las)
}   
```
and then `track_sensor()`:

 ```r
track_sensor(ctg, Roussel2020(), multi_pulse = T, thin_pulse_with_time = 0)
#> Processing [=================>-------------------------------------------------------------------------------------------------------------------------------]  12% (1/8) eta: 22sError: After keeping only first and last returns of multiple returns pulses, 11 pulses still have more than 2 points. This dataset is corrupted and gpstime is likely to be invalid.
las <- readLAS(catalog_makechunks(ctg)[[1]])
las@data[, prob:= .N > NumberOfReturns, by = c("gpstime", "UserData")]
head(las@data[prob == T, ], 1)$gpstime
#> [1] 208541938.6586057841778
> las@data[gpstime == 208541938.6586057841778]
#>                          X                       Y                       Z                 gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline
#> 1: 362899.7500000000000000 4727392.240000000223517 124.4600000000000079581 208541938.6586057841778      1416            1               1                 0                0
#> 2: 363099.8300000000162981 4727178.900000000372529 157.8799999999999954525 208541938.6586057841778       528            2               4                 0                0
#> 3: 363099.1799999999930151 4727178.799999999813735 154.5500000000000113687 208541938.6586057841778       818            3               4                 0                0
#> 4: 363096.9700000000302680 4727178.450000000186265 143.3499999999999943157 208541938.6586057841778       843            4               4                 0                0
#>    Classification ScannerChannel Synthetic_flag Keypoint_flag Withheld_flag Overlap_flag                ScanAngle UserData PointSourceID buffer  prob
#> 1:              1              1          FALSE         FALSE         FALSE        FALSE -16.99800109863281250000        0            50      0  TRUE
#> 2:              1              2          FALSE         FALSE         FALSE        FALSE -10.99800014495849609375        0            50      4 FALSE
#> 3:              1              2          FALSE         FALSE         FALSE        FALSE -10.99800014495849609375        0            50      4 FALSE
#> 4:              2              2          FALSE         FALSE         FALSE        FALSE -10.99800014495849609375        0            50      4 FALSE
```

Let's call the file where `buffer = 0` file A, and the file where `buffer = 4` file B...
This `gpstime` group isn't split by `UserData` despite having two groups of `NumberOfReturns`. This is because one half of the group (points 2, 3, and 4) is in file B and this would have been flagged as `UserData = 1` **but** we don't write to this file since it is the buffer. When file B is processed we flag the same group (points 2, 3, and 4) as `UserData = 0` and thus we have a problem. That is why we need to jump through hoops to incorporate the buffer in our splitting. Otherwise we get the scenario where both tiles write `UserData = 0` for their half of the group.  

You wouldn't expect this to happen because we are sorting by `NumberOfReturns` but what this tells me is that point 1 (in the above set of data) does not fall in the buffer of file B despite points 2, 3, and 4 being in the buffer of file A...It's quite confusing and it's caused me to scratch my head several times.

Jean-Romain, 2020-11-26:
I think I understand. Point 1 (pulse 1) is far from points 2,3,4 (pulse 2). When reading file A you load pulse 1 and pulse 2 because pulse 2 is close to the edge but when reading file B, pulse 2 is loaded but not pulse 1 because pulse 1 is far from the edges.

I guess the edges is at X = 363000. Pulse 1 is 100.25 m far from the edges, pulse 2 is 99.75 m from the edges, with a buffer of 100 m it fails. But with a buffer of 101m it should work. But if you set a buffer of 101 m you will have a problem with a pulse at 102 m and so on...

But:
- But this is a problem only if you want to write the information in files. If you compute the user data on-the-fly before to run `track_sensor` this no longer an issue.
- You can use the ScanAngle as an additional information to retrieve the beams (may resolve the problem with pulse with the same NumberOfReturn)

Jean-Romain, 2020-11-26:
You can also work by flightline.

```r
las = readLAS(ctg, select = "xyzcarn", filter = "-drop_single -keep_point_source xxx")
las = indentify_beams(las)
sensor = track_sensor(las)
```

Then you loop on each flightline id to compute each track.

lucas-johnson, 2020-11-26:
Ahhh I see now. If we compute on the fly – as part of a custom track_sensor function we can avoid this problem. That makes sense now. Using `ScanAngle` probably makes sense as well – I'll give this a try. 

I understand how we might loop on flightline, but I'm not so sure how that would be effective – mostly because I don't understand how flightline relates to pulse/beam or buffers.

Thanks for your suggestions and working through this with me. Very helpful.

Jean-Romain, 2020-11-26:
No buffer no LAScatalog, you manually load the whole flighlines one by one. If you load only multiple return `-drop_single` you wont have too much points. You catalog is not that big

closed 2020-11-26 by Jean-Romain