r-lidar/lidR #398: Warning from future when plan(multisession) set

quanl88, 2020-11-29:
The code below to compute a normalized las file produces the following warning when `plan(multisession)` is set. Since `seed=TRUE` is an argument of `future` and not `plan`, I don't know how to ensure that it is set when `future()` is invoked by `normalize_height`.

```
library(lidR)
library(future)
plan(multisession, workers=2L)

cat<-readLAScatalog("C:\lidar\")

opt_output_files(TFL)<-"C:\norm\{ORIGINALFILENAME}_norm"

normalize_height(TFL, knnidw())
```

```
UNRELIABLE VALUE: Future (‘<none>’) unexpectedly generated random numbers without
specifying argument '[future.]seed'. There is a risk that those random numbers are not
statistically sound and the overall results might be invalid. To fix this, specify argument
'[future.]seed', e.g. 'seed=TRUE'. This ensures that proper, parallel-safe random numbers
are produced via the L'Ecuyer-CMRG method. To disable this check, use
[future].seed=NULL, or set option 'future.rng.onMisuse' to "ignore".
```

Note, the same warning is produced when computing `grid_metrics` with `plan(multisession)` set.

Thank you for your continuing support of lidR.

Jean-Romain, 2020-11-29:
Thank you for reporting. This is a new but known issue that appeared with the latest version of `future` . No need to set `multisession` to reproduce actually. It is a little bit trickier to fix than expected. See https://github.com/HenrikBengtsson/future/issues/446 for more details.

For now you can use 

```r
options(future.rng.onMisue = "ignore")
```

to ignore the warning. No random number are expected to be generated. I'm currently investigating where the hell it comes from. Likely not from `lidR`. Maybe from `raster`...

Jean-Romain, 2020-11-30:
I found 1 occurrence of unexpected RNG in `catalog_apply()`

```r
if (interactive()) {
  invisible(addTaskCallback(local({
    last <- .GlobalEnv$.Random.seed
    
    function(...) {
      curr <- .GlobalEnv$.Random.seed
      if (!identical(curr, last)) {
        msg <- "TRACKER: .Random.seed changed"
        if (requireNamespace("crayon", quietly=TRUE)) msg <- crayon::blurred(msg)
        message(msg)
        last <<- curr
      }
      TRUE
    }
  }), name = "RNG tracker"))
}

# Only with .tif
library(raster)
tmp <- tempdir()
r <- raster(system.file("external/test.grd", package="raster"))
rf <- writeRaster(r, filename=file.path(tmp, "allint.tif"), datatype='INT4S', overwrite=TRUE)
```

Jean-Romain, 2020-12-07:
fixed

closed 2020-12-07 by Jean-Romain

quanl88, 2020-12-09:
Thanks, Jean-Romain.
This was fixed in RGDAL?

Jean-Romain, 2020-12-09:
No I just set `seed = TRUE` by default in `future`