r-lidar/lidR #399: Lascatalog ErrorFilter

diouck, 2020-12-03:
Hello ,
Here is my problem: I have data, when I import a tired file and I put these filters there is no problem.
`las <- readLAS("data/Nuage3D2018_1353800_6235000.laz", filter = "-keep_class 2 3 4 5 19 -drop_z_below 2 -drop_z_above 50", select = "xycrni") 
`
 But because when I go through lascatalog, impossible to filter the data
`opt_chunk_size(ctg) <- 350
opt_chunk_buffer(ctg) <- 350
opt_select(ctg) <- "xycrni"
opt_filter(ctg) <- "-keep_class  2 3 4 5 19 -drop_z_below 2 -drop_z_above 50"  

after filter

class       : LAScatalog (v1.4 format 7)
extent      : 1353800, 1355200, 6235000, 6236000 (xmin, xmax, ymin, ymax)
coord. ref. : NA 
area        : 1.4 kunits²
points      : 7.85 million points
density     : 5.6 points/units²
num. files  : 1

before

class       : LAScatalog (v1.4 format 7)
extent      : 1353800, 1355200, 6235000, 6236000 (xmin, xmax, ymin, ymax)
coord. ref. : NA 
area        : 1.4 kunits²
points      : 7.85 million points
density     : 5.6 points/units²
num. files  : 1

 `

 are you an idea or  how to take these filters into account thank you

Jean-Romain, 2020-12-03:
The filters only apply *at read time* when processing (i.e. when `readLAS()` is called). The LAScatalog only reads and displays the metadata of the files and cannot know how many points fit your selection without reading every single file which may take a significant amount of time.

If you have more questions please ask on https://gis.stackexchange.com/questions/tagged/lidr with the tag `lidr`

closed 2020-12-03 by Jean-Romain

JakubCha, 2020-12-03:
Hey,

opt_filter is taken into account during processing of LASCatalog. You may consider it "lazy loading". When you call catalog what you actually see is LAS files metadata.