r-lidar/lidR #400: clip with catalog and output_files breaks when no points are found

lucas-johnson, 2020-12-07:
Clipping a catalog works despite no points being found (i.e. no error), however throws an error if `opt_output_files()` is set. I don't think this should create an error, rather no files should be written.
 
```r
clip_roi(ctg, las_plots)
#> No point found for within region of interest 1.
#> class        : LAS (v1.4 format 6)
#> memory       : 13.2 Kb 
#> extent       : 0, 0, 0, 0 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NA 
#> area         : 0 units²
#> points       : 0  points
#> density      : 0 points/units²
opt_output_files(ctg) <-  "~/{RGID}"
clip_roi(ctg, las_plots)
#> No point found for within region of interest 1.
#> Error: written_path is not character it is a NULL
```

closed 2020-12-07 by Jean-Romain

Jean-Romain, 2020-12-07:
Congrats you opened the 400th issue!

Fixed. Thanks for reporting

lucas-johnson, 2020-12-07:
Thanks for fixing!

commit d56fd0cc (2020-12-07): Fix #400