r-lidar/lidR #401: Question (Not Issue) with LAScatalog functionality

zhilgy92, 2020-12-07:
Hello! I am not sure if the is the best location to post questions, but I am trying to work through a few things, using the lidR package and was wondering if I could ask advice here. I've looked through the lidR book and, while an incredible resource, I cannot figure out my issue there, either.

I have a tiled LAS dataset from an aerial LiDAR collection at a coastal site in British Columbia. I have used the LAScatalog to successfully combine tiles. I would like to combine the tiles and export them as a single point cloud. My questions then are:
1) Is this possible? Am I able to export the catalog as a single las point cloud? I need to bring the cloud into a software like CloudCompare, to ensure alignment with other point clouds from other surveys, and this is not easy to do with 40 different tiles.
2) If possible, do the combined clouds retain their classifications (return #, code, etc.) when combined and exported? 

If this is not the best place to ask these questions, I apologize! I appreciate your help and look forward to hearing back from you, either way!

Jean-Romain, 2020-12-08:
Hi. I no longer answer to questions about lidR by email or on github because I receive too many questions from too many users. If you have questions about using the lidR package please ask on [gis.stackexchange.com](https://gis.stackexchange.com/) with the tag `lidr`. On stackexchange.com other users can also answer your questions and the answers thus provided will be public and well indexed by search engines so they can also help other users. Moreover we can provide link to already answered questions.

That being said, yes you can but I don't think it is a good idea. Will give you a detailed answer on stack exchange.

closed 2020-12-08 by Jean-Romain

zhilgy92, 2020-12-08:
Thank you for letting me know! I have posted the question, with the lidR
tag!