r-lidar/lidR #409: lidR Error in st_crs.character(comment(x))

malonzo47, 2021-02-17:
I load this las file (https://drive.google.com/file/d/1y69AHcuRNFbknTYuYa7yHWEninApTKwi/view?usp=sharing) using readLAS and then do print(las). I get this error:

Error in st_crs.character(comment(x)) : 
  invalid crs: PROJCRS["WGS 84 / UTM zone 6N",
    BASEGEOGCRS["WGS 84",
        DATUM["World Geodetic System 1984",
            ELLIPSOID["WGS 84",6378137,298.257223563,
                LENGTHUNIT["metre",1]]],
        PRIMEM["Greenwich",0,
            ANGLEUNIT["degree",0.0174532925199433]],
        ID["EPSG",4326]],
    CONVERSION["UTM zone 6N",
        METHOD["Transverse Mercator",
            ID["EPSG",9807]],
        PARAMETER["Latitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8801]],
        PARAMETER["Longitude of natural origin",-147,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8802]],
        PARAMETER["Scale factor at natural origin",0.9996,
            SCALEUNIT["unity",1],
            ID["EPSG",8805]],
        PARAMETER["False easting",500000,
            LENGTHUNIT["metre",1],
            ID["EPSG",8806]],
        PARAMETER["False northing",0,
            LENGTHUNIT["metre",1],

Note: this las file may look a little odd since it's an SfM point cloud but it generally works fine.

Thanks,
Mike

Jean-Romain, 2021-02-17:
Tested but not reproduced:

```
library(lidR)
las = readLAS("~/issue 409/201807XX_Denali_FHM_mid_group1_densified_point_cloud_part_3.las")
las
#> class        : LAS (v1.2 format 3)
#> memory       : 26.8 Mb 
#> extent       : 646776.9, 646922.4, 6961000, 6961092 (xmin, xmax, ymin, ymax)
#> coord. ref.  : WGS 84 / UTM zone 5N 
#> area         : 8318.014 m²
#> points       : 305.3 thousand points
#> density      : 36.71 points/m²
```

Please give more details on your setup

- R, lidR, rlas, sp, sf, rgdal versions
- OS
- output of `rgdal:::rgdal_extSoftVersion()`

malonzo47, 2021-02-17:
R version 3.6.2 (2019-12-12)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19041)
rgdal_1.5-23
raster_3.3-13
sp_1.4-5
sf_0.9-1

   GDAL GDAL_with_GEOS           PROJ             sp 
       "3.2.1"         "TRUE"        "7.2.1"        "1.4-5"

Jean-Romain, 2021-02-17:
Tested in my virtual machine on R 3.6.2 and everything was ok. 

You did not mentioned the versions of `lidR` and `rlas`. Anyway at this stage my best advice is to move on with R 4.0. R 3.6 is two years old and uses an old toolchain (old C++ compiler) and recent versions of lidR are not available for Windows for R 3.6. Even if your issue is not related to the R toolchain, R 4.1 will be released soon and new package updates will no longer be available for R 3.6. It is time to say bye bye to out-of-date softwares.

If moving to R 4.0 does not solve your issue please tell me. If the problem is fixed tell me too.

malonzo47, 2021-02-17:
Ok, a bit painful to let go of 3.6 but I'm not up on R 4.0. With the new setup the problem is resolved. Thanks for your help.

closed 2021-02-17 by malonzo47