r-lidar/lidR #411: Multisession and las_check on LASCatalog

JulienPoint, 2021-03-15:
Hi again Jean-Romain,

When a attempt to run a las_check() on a LASCatalog with parallelization using future package : 
```
library(lidR)
library(future)

plan(multisession)

ctg <- readLAScatalog(path_to_folder")
checks <- las_check(ctg, print = FALSE, deep = TRUE)
```

I get the following error :
```Erreur : pas de méthode pour 'las_check' applicable pour un objet de classe "c('LAScluster', 'Spatial')"```

Furthermore, print=TRUE seems to double print the output. To confirm.

Julien

Jean-Romain, 2021-03-15:
Thank you for reporting. I fixed the error with parallel processing. I'm investigating on the double print.

JulienPoint, 2021-03-15:
You're welcome. I just tested the commit, it's perfectly working.
I take the opportunity to ask : would it be possible to obtain, in addition to the compact and useful list, an equivalent of the printed output when storing the `las_check()` result? It would be cleaner than catching the console output!

Jean-Romain, 2021-03-15:
It is already the case but it is returned invisibly (it is returned visibly if `print = FALSE`). See the doc

> ### Value
> A list with two elements named warnings and errors. This list is returned invisibly if print = TRUE. If deep = TRUE a nested list is returned with one element per file.

```r
u = las_check(ctg, deep = TRUE)
u 
#> $Megaplot.laz
#> $Megaplot.laz$warnings
#> [1] "'PointSourceID' attribute is not populated."    "'EdgeOfFlightline' attribute is not populated."
#> $Megaplot.laz$errors
#> character(0)
#> 
#> $MixedConifer.laz
#> $MixedConifer.laz$warnings
#> [1] "1 points are duplicated and share XYZ coordinates with other points"  "'PointSourceID' attribute is not populated."     "'ScanDirectionFlag' attribute is not populated." "'EdgeOfFlightline' attribute is not populated." 
#> $MixedConifer.laz$errors
#> [1] "The following attributes contain NAs: treeID" "ESPG code and proj4string do not match."     
#> 
#> $Topography.laz
#> $Topography.laz$warnings
#> [1] "'ScanDirectionFlag' attribute is not populated." "'EdgeOfFlightline' attribute is not populated." 
#> $Topography.laz$errors
#> [1] "ESPG code and proj4string do not match."
```

I'm still unable to figure out why the double print. But I'll eventually fixed it but probably not today.

Jean-Romain, 2021-03-15:
Noticed that I changed the behaviour of `las_check` (not committed on github yet). There are now three types of message ("messages", "warnings", "errors"). For example *"'PointSourceID' attribute is not populated."* is now a message instead of a warning.

JulienPoint, 2021-03-15:
Ok I wasn't sure about it, so here's my experience : for example, for files without normalization ("no" appears in the console output, see below) I can't see a record, neither within `$warnings` nor `$errors`.

```
u = las_check(ctg, deep = TRUE)

____________________________________
Checking the file: sample_points.laz
____________________________________

 Checking the data
  - Checking coordinates... <U+2713>
  - Checking coordinates type... <U+2713>
  - Checking coordinates range... <U+2713>
  - Checking coordinates quantization... <U+2713>
  - Checking attributes type... <U+2713>
  - Checking ReturnNumber validity... <U+2713>
  - Checking NumberOfReturns validity... <U+2713>
  - Checking ReturnNumber vs. NumberOfReturns... <U+2713>
  - Checking RGB validity... <U+2713>
  - Checking absence of NAs... <U+2713>
  - Checking duplicated points...
   <U+26A0> 3569 points are duplicated and share XYZ coordinates with other points
  - Checking degenerated ground points... <U+2713>
  - Checking attribute population... <U+2713>
  - Checking gpstime incoherances <U+2713>
  - Checking flag attributes... <U+2713>
  - Checking user data attribute... <U+2713>
 Checking the header
  - Checking header completeness... <U+2713>
  - Checking scale factor validity... <U+2713>
  - Checking point data format ID validity... <U+2713>
  - Checking extra bytes attributes validity... <U+2713>
  - Checking the bounding box validity... <U+2713>
  - Checking coordinate reference sytem... <U+2713>
 Checking header vs data adequacy
  - Checking attributes vs. point format... <U+2713>
  - Checking header bbox vs. actual content... <U+2713>
  - Checking header number of points vs. actual content... <U+2713>
  - Checking header return number vs. actual content... <U+2713>
 Checking preprocessing already done 
  - Checking ground classification... yes
  - Checking normalization... no
  - Checking negative outliers... <U+2713>
  - Checking flightline classification... yes
```

``` r
u
$sample_points.laz
$sample_points.laz$warnings
[1] "3569 points are duplicated and share XYZ coordinates with other points"

$sample_points.laz$errors
character(0)
```

JulienPoint, 2021-03-15:
> Noticed that I changed the behaviour of `las_check` (not committed on github yet). There are now three types of message ("messages", "warnings", "errors"). For example _"'PointSourceID' attribute is not populated."_ is now a message instead of a warning.

Oh ok, maybe the _normalization_  check will fall under messages.

Jean-Romain, 2021-03-15:
Well, no. Actually the information about the normalization is neither a message nor a warnings and is not in the output. But I guess it can be seen as a feature request. Please open an new issue with a feature request.

JulienPoint, 2021-03-15:
<U+2713> comes along with the `deep=TRUE` with a LASCatalog.
No problems with `las_check(ctg, print = TRUE, deep = FALSE)` or with `las_check(single_las)`

``` r
sessionInfo()
R version 4.0.4 (2021-02-15)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19041)

Matrix products: default

locale:
[1] LC_COLLATE=French_France.1252  LC_CTYPE=French_France.1252    LC_MONETARY=French_France.1252
[4] LC_NUMERIC=C                   LC_TIME=French_France.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] tictoc_1.0    future_1.21.0 lidR_3.1.3    raster_3.4-5  sp_1.4-5     

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.6         parallelly_1.24.0  magrittr_2.0.1     units_0.7-0        lattice_0.20-41    globals_0.14.0    
 [7] tools_4.0.4        parallel_4.0.4     grid_4.0.4         data.table_1.14.0  KernSmooth_2.23-18 e1071_1.7-4       
[13] DBI_1.1.1          rgeos_0.5-5        rlas_1.3.9         class_7.3-18       lazyeval_0.2.2     digest_0.6.27     
[19] crayon_1.4.1       sf_0.9-7           codetools_0.2-18   glue_1.4.2         compiler_4.0.4     classInt_0.4-3    
[25] listenv_0.8.0
```

closed 2021-03-15 by Jean-Romain