r-lidar/lidR #412: las_check() prints and results consistency

JulienPoint, 2021-03-15:
Hi,

In response to this [post](https://github.com/Jean-Romain/lidR/issues/411#issuecomment-799464012_), would it be possible to have all information printed by the `las_check()` output available in the list returned by the function?

For example :

- _normalization_ status is printable with `deep=TRUE` on a `LASCatalog`, but not stored in the list   
- _indexation_ is checked and printable for a `LASCatalog `(`deep = FALSE`), but not with `deep = TRUE `or when running `las_check()`on a single file, and does not appear in the list.

Jean-Romain, 2021-03-15:
> indexation is checked and printable for a LASCatalog (deep = FALSE), but not with deep = TRUE or when running las_check()on a single file, and does not appear in the list.

`deep = TRUE` almost is equivalent to:

```r
for (file in files) {
  las = readLAS(file)
  las_check(las)
}
```

And there is no check of indexation for LAS objects because they are not associated to a file. It only checks for file indexation.

> normalization status is printable with deep=TRUE on a LASCatalog, but not stored in the list

I added new messages in the output of `las_check()` both for `LAS` and `LAScatalog`

```r
las_check(las, print = FALSE)
#> $messages
#>  [1] "'PointSourceID' attribute is not populated."    "'EdgeOfFlightline' attribute is not populated." "The point cloud is ground classified"          
#> $warnings
#>  character(0)
#> $errors
#>  character(0)

las_check(ctg, print = FALSE)
#> $messages
#>  [1] "The LAS files are not spatially indexed"
#> $warnings
#>  character(0)
#> $errors
#>  character(0)
```

JulienPoint, 2021-03-15:
> And there is no check of indexation for LAS objects because they are not associated to a file. It only checks for file indexation.

Yes I did not realized that.

> I added new messages in the output of `las_check()` both for `LAS` and `LAScatalog`

Looks perfect to me, thanks!

closed 2021-03-15 by JulienPoint