r-lidar/lidR #415: replace aoi in an existing LAScatalog, with a new LAScatalog for subsequent processing

rs806, 2021-03-18:
I would like to have a method to replace a section of old LAS files with new LAS files, while taking advantage of lidR buffering around the edge of the new LAS files extent. Something like catalog_intersect but instead of selecting it would replace plus select adjacent old files for buffering purposes.

Jean-Romain, 2021-03-18:
Sounds too specific as is. Especially what you are asking can only work if you try to replace old tiles with new tiles that have the exact same bounding boxes. Could we imagine a more generic tools such as `rbind` for `LAScatalog`? Or two more generic functions that, when used successively, can achieve this task ?

rs806, 2021-03-18:
For our immediate work, extents of new tiles match old tiles exactly. But you're point is good as this won't be the case in the future (say if lidar is collected by an RPAS). I'm having trouble conceptualizing the steps, however I wonder if, then, it is possible to create an exclusion mask (maybe by using `catalog_boundaries()` from the catalog of the new data. That mask could be overlaid on the old catalog to crop or to flag points for exclusion using `filter_poi()` and the rest would have the buffer value. Then use `rbind` to combine the new and old catalog.

Jean-Romain, 2021-03-18:
It is definitively complex and highly specific. Basically you want to add tiles to a LAScatalog that will be used as buffer only but those tiles may overlap partially the original collection.  This will require an in depth modification of the engine so it is a no go option.

In the case of non-overlapping tiles however it is easier and doable geometry operations using `sf`. In the following I'm using 3 collections. The blue is the main one I want to process, the red and green ones are the ones I want to use as buffer only. Lets call them main collection and secondary collections. The 3 collections are not exactly complementary. If the collections were exactly complementary we could skip the geometry operations.

*Notice that the blue catalog contains duplicated tiles which is a problem but does not matter here for the example*

``` r
library(lidR)
library(sf)

ctg1 = readLAScatalog("lidar1/", recursive = TRUE)
ctg2 = readLAScatalog("lidar2/", recursive = TRUE)
ctg3 = readLAScatalog("lidar3/", recursive = TRUE)

plot(ctg1)
plot(ctg2, col = rgb(1,0,0,0.5), add = T)
plot(ctg3, col = rgb(0,1,0,0.5), add = T)
```

[image]

We can now remove the duplicated tiles that are found both in the main collection and the secondary collections

``` r
sf1 <- st_as_sf(ctg1)
sf2 <- st_as_sf(ctg2)
sf3 <- st_as_sf(ctg3)

sf1 <- st_union(st_buffer(sf1, 0.01))
id2 <- st_contains(sf1, sf2)[[1]]
id3 <- st_contains(sf1, sf3)[[1]]

ctg2 <- ctg2[-id2,]
ctg3 = ctg3[-id3,]

plot(ctg1)
plot(ctg2, col = rgb(1,0,0,0.5), add = T)
plot(ctg3, col = rgb(0,1,0,0.5), add = T)
```

[image]

At this stage we can flag the main collection as `processed = TRUE` and secondary collection as `processed = FALSE` then bind the collections. And we can see that we have the expected collection (minus the duplicated tiles that must be removed in the main collection)

``` r
ctg1$processed = T
ctg2$processed = F
ctg3$processed = F

ctg1@data = rbind(ctg1@data, ctg2@data, ctg3@data)
plot(ctg1)
```

[image]

And we are done!! 

```r
plot(ctg1, chunk = T)
```
[image]

:warning:  :warning: **But but but...**. We have a problem. In my case the collections are not compatible. You can see that there are both LAS file formats 1 and 3. At the edges you will need to read files in format 1 and 3 as a single point cloud and your script will crash because they cannot be read together.

```r
ctg1
#> class       : LAScatalog (v1.2 format 1 and 3)
#> extent      : 324000, 360000, 5304000, 5331000 (xmin, xmax, ymin, ymax)
#> coord. ref. : NAD83(CSRS) / MTM zone 8 
#> area        : 1027.98 km²
#> points      : 7.37 billion points
#> density     : 7.2 points/m²
#> num. files  : 1028 
```

I already processed such collection actually this is why I build the example pretty easily but I had to modify the source code temporarily. Actually it is doable to read different files format at once but in the current state it is not because I did not spent time coding all the safety guard it requires to be made public. I did it for myself because I knew was I was doing.

rs806, 2021-03-18:
Great, thanks Jean-Romain.  I have yet to try this on my own data with equal LAS tile extents but will have a chance soon. In the hypothetical future case where I might integrate an RPAS lidar survey, I would want to direct the survey to ensure a buffer large enough to encompass a reasonable tiling scheme. Say if the RPAS lidar is delivered in complete 250 metre tiles, I could still integrate this into ALS lidar which is at 1,000 metre tiles by using `catalog_retile()`.

Jean-Romain, 2021-03-18:
With `catalog_retiles()` you can do everything. But it is a lot of work.

I added a function `rbind` for `LAScatalog` in version 3.2.0 available in `devel` branch. It warns about potential problems by running `las_check()` internally. With `rbind()` + my previous answer you have all you need but I won't create any highly specific tool. So I'm closing this issue. Thanks

```r
ctg = rbind(ctg1, ctg2, ctg3)
#> Warning messages:
#> 1: Inconsistent scale factors 
#> 2: Inconsistent offsets 
#> 3: Some tiles seem to overlap each other 
#> 4: Inconsistent CRS accross files 
```

closed 2021-03-18 by Jean-Romain