r-lidar/lidR #419: LAD function

Saadi4469, 2021-03-30:
Hello,

I am new to the LAD function, so I would appreciate some guidance. How can one determine the "dz" and the "k" values to be used for a particular LAS file?

Thank you,

Saadi4469, 2021-03-30:
Or dz = 1 and k = 0.5 are the default values to be used?

Jean-Romain, 2021-03-30:
Did you read the documentation? Did you read the paper that is cited in the documentation?

Saadi4469, 2021-03-30:
Yes, I did and the authors do mention that k is well approximated by 0.5. However, the paper states dz as given thickness and doesn't say how to calculate or find this value, and the documentation in R states it as layer thickness (height bin). So I don't really know how to find dz.

Jean-Romain, 2021-03-30:
So, you know as much as I do. I'm making the tools to reproduce the state of the art but can't tell you what you should or should not do with that. It is up to you to take good decisions.  I can only help on *how to" questions. If you really want advices on "what to do" you should ask somebody else because I don't have any authority to tell anybody what they should or should not do.

Regards

closed 2021-03-30 by Jean-Romain

Saadi4469, 2021-03-30:
@hriihimaki thank you for your guidance, unfortunately this is part of a tutorial, so all I know about this LAZ file is that it's from somewhere in Alaska. And, I don't have any ground truth data or even satellite imagery of this area to know more about the vegetation type. The application however is to simply plot LAD. Hope this makes sense. Any further guidance on this will be much appreciated.

Cheers,

Saadi4469, 2021-03-30:
Thank you for sharing this, I used the following code to plot LAD, and I know it's hard to tell whether it's right or wrong without knowing about the area.

```
lad = LAD(Als_P6_Sub_Ori@data$Z,dz = 1,k = 0.5)
plot(lad, type="l", xlab="Elevation", ylab="Leaf area density", main = "Leaf Area Density")
```

[https://imgur.com/GU64MZo](url)

Jean-Romain, 2021-03-30:
your data is not height normalized so I'd say it's wrong

Saadi4469, 2021-03-30:
Thank you for pointing it out. I have revised my code and I also computed LAD from segmented trees using zmax as tree height. Would this also be correct?

lad = LAD(las_normalized_1@data$Z,dz = 1,k = 0.5)
plot(lad, type="l", xlab="Height", ylab="Leaf area density", main = "Leaf Area Density")

Result:
https://imgur.com/RtBvfOS   

Another approach

lad_1 = LAD(hulls@data$zmax,dz = 1,k = 0.5) # Hulls were computed using tree_hulls after segmenting trees using Silva2016 once the LAS file was height normalized.

Result:
https://imgur.com/UlhbLTO