r-lidar/lidR #423: Grid_metrics- possible artifacts from differences in point density?

jason-drake, 2021-04-15:
Hello Jean-Romain,
First off, I want to thank you for all your years of hard work to create, maintain and update such an incredible package!  

Our lab has been using lidR for about a year and have noticed some occasional "artifacts" in the raster products from grid_metrics. Specifically, we are seeing linear areas with much lower values for some metrics (for example the zq50 from stdmetrics_z shown below). 
[image]

This issue seems to occur when there is a sharp change in point density (see corresponding grid_density below).  
[image]

I really appreciate any insight or help that you can provide because I cannot pinpoint what is causing this issue or how to solve it. I have done my best to share this as a minimal reproducible example but if there is any additional information you need please let me know. 

Here is the minimal code:
```{r}
library(lidR)
las <- readLAS("C:/tmp/e1050n0849.las", , filter="-drop_withheld")
stmet = grid_metrics(las, .stdmetrics_z, res = 5)
col <- height.colors(50)
plot(stmet[[18]], col=col)  #50th percentile of height distribution
gdens<- grid_density(las, res=5)
plot(gdens)
```

And the LAS file can be downloaded [here.](https://drive.google.com/file/d/1ZwJMPh4OM6po-9TuKk0EARlxK-ODi5KI/view?usp=sharing)

Jean-Romain, 2021-04-15:
There is no way it is an algorithm artefact. `grid_metrics()` does its job of taking points in each pixel and computing "something" on these points. Clearly your "problem" is correlated to the locally high density (likely a flightline overlaps). So there is something with your point cloud and it is you responsibility to know what the data provider gave you. But I can make  a quick guess. I think it is related to point flagged `Withheld`. You can see you have many of them (why? it seems to be related to scan angle). By removing them you have an unnatural edge.

```r
library(lidR)
las <- readLAS("e1050n0849.las", select = "xyzw")
stmet  grid_metrics(las, sum(Withheld_flag), res = 5)
col <- height.colors(50)
plot(stmet, col=col) 
```
[image]

Using the `Withheld` points your artefact moved to the new edges of the flightlines

```r
library(lidR)
las <- readLAS("e1050n0849.las", select = "xyz")
stmet = grid_metrics(las, .stdmetrics_z, res = 5)
col <- height.colors(50)
plot(stmet[[18]], col=col)
```

[image]

Now it is between you and your data provider to know which points must be discarded or not.

jason-drake, 2021-04-16:
Thanks Jean-Romain, I really do appreciate your help and advice! These datasets were provided to USGS in 2018 so unfortunately we are not able to work directly with the vendor to resolve this issue. We will look closely into the withheld points and scan angle information to see if there are other options to filter out points that may be causing this issue. Interestingly it seems to be primarily impacting height percentiles in the middle of the distribution (zq35-zq70)  but not in the lower or upper range.  

If you or any other users have any additional QA/QC techniques that might be helpful to diagnose this issue please let me know.
Thanks!

Jean-Romain, 2021-04-16:
Look carefully at the right and left edges in both cases:

```r
las <- readLAS("~/Téléchargements/e1050n0849.las", select = "xyzp", filter = "-keep_random_fraction 0.1")
plot(filter_poi(las, PointSourceID == 11903, Z < 55))
plot(filter_poi(las, PointSourceID == 11904, Z < 55))

las <- readLAS("~/Téléchargements/e1050n0849.las", select = "xyzp", filter = "-drop_withheld -keep_random_fraction 0.1")
plot(filter_poi(las, PointSourceID == 11903, Z < 55))
plot(filter_poi(las, PointSourceID == 11904, Z < 55))
```

[image]

You have your answer. This is an effect of scan angle. They should have flagged withheld points strait. Here you have a geometric artefact at the edges of flightline. But not only because not mater which angle you used as threshold you have this necessarily this artefact

```r
plot(cl, color = "ScanAngle", col = pastel.colors(20))
```

[image]

Jean-Romain, 2021-04-16:
It the first time I see an effect so strong. I might be because the flightlines are particularly strait. When flightline are not straight the effect is reduced and not visible because does not create strait pattern. Moreover we rarely computer median with a resolution of 5m

closed 2021-04-16 by Jean-Romain

jason-drake, 2021-04-16:
Thank you very much Jean-Romain! This is incredibly helpful!